sandeepmistry/arduino-nRF5 #84: Uart::IrqHandler() can lose bytes if interrupted (nRF52)

KyleKubik, 2017-01-22:
Apologies if this is not the correct format, I don't use GitHub much.

I've verified this issue on my nRF52832 chip, I'm not sure if it will act in the same way on other chips.

The current code for Uart::IrqHandler() is

```
void Uart::IrqHandler()
{
  if (nrfUart->EVENTS_RXDRDY)
  {
    rxBuffer.store_char(nrfUart->RXD);

    nrfUart->EVENTS_RXDRDY = 0x0UL;
  }
}
```

However, the issue with this is that the byte is read out of the UART buffer before the RXDRDY flag is cleared, meaning if a new byte arrives before store_char() is finished (due to a higher priority interrupt), that new byte will be "lost" due to the code clearing the flag after it has arrived.

The simple fix to this issue is to simply swap the order of the two lines, like so:

```
void Uart::IrqHandler()
{
  if (nrfUart->EVENTS_RXDRDY)
  {
    nrfUart->EVENTS_RXDRDY = 0x0UL;

    rxBuffer.store_char(nrfUart->RXD);
  }
}
```

This may not be the case for all nRF5 chips, but it is at least for the nRF52832. More info in the product specification: http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.sds%2Fdita%2Fsoftdevices%2Fs130%2Fprocessor_avail_interrupt_latency%2Fprocessor_avail_interrupt_latency.html

sandeepmistry, 2017-01-30:
@KyleKubik do you have a way to reproduce this condition so that we can test before and after your suggested change?

sandeepmistry, 2017-02-13:
Closing for now due to lack of activity.

closed 2017-02-13 by sandeepmistry

mohsenemteq, 2018-01-02:
Swapping these lines solved my problem. I had an issue where when I tried receiving UART data on a regular bases the UART would freeze. I have changing these two lines now I have UART receiving 20 bytes every 20 ms and passing it to a phone through BLE

mohsenemteq, 2018-01-02:
Swapping these lines solved my problem. I had an issue when I tried receiving UART data on a regular bases the UART would freeze. I have changing these two lines now I have UART receiving 20 bytes every 20 ms and passing it to a phone through BLE

sandeepmistry, 2018-01-07:
@mohsenemteq @KyleKubik please open a pull request for these changes. As well as include a test sketch to reproduce the condition and verify the fix.