sandeepmistry/arduino-nRF5 #88: DS1307 not recognised.

szymonm2, 2017-01-31:
As the title suggests the ds1307 isnt recognised by the nrf51822. I think i did not set the i2c pins? how do i do that? I tried using "Wire.begin(4,5);" but it told me that it is invalid. I also tried looking up in google but no luck.

dlabun, 2017-02-02:
The I2C bus functions the same as any of the official Arduino boards, you can find more information on the Arduino site:

https://www.arduino.cc/en/Reference/WireWrite

szymonm2, 2017-02-02:
> The I2C bus functions the same as any of the official Arduino boards

Im not sure about the pinout though. what pins are used for sda and scl on the nrf51822. On the arduino you have dedicated ones (A4,A5). what about the nrf51822. I cannot find any information about this

dlabun, 2017-02-02:
It all depends on your board... Are you using one of the supported boards or are building your own?

szymonm2, 2017-02-02:
> Are you using one of the supported boards or are building your own
I am using the BLE400 nrf51822 board but without the development add on

dlabun, 2017-02-02:
I'm not sure what you mean by that as the BLE400 is a development board

szymonm2, 2017-02-02:
``BLE400 is a development board"

the pcb that plugs into the ble400

dlabun, 2017-02-02:
Do you mean the Core51822 module?

szymonm2, 2017-02-02:
> Core51822 module?

yes

dlabun, 2017-02-02:
You have a few different options then... Figure out the pin out of the BLE400 motherboard and use the same pins on the Core51822 or you can create your own variant and specify what pins you'd like.

szymonm2, 2017-02-02:
>  specify what pins you'd like.

how do i do that. I know what pins are what, but i have no idea how to assign the i2c pins to a given pin

dlabun, 2017-02-02:
If you need help modifying an existing variant or creating a new one I would recommend you consult the Arduino forums for more information... This is more of a generic Arduino question that can be answered a lot faster and better on the forum.

sandeepmistry, 2017-02-06:
@dlabun thanks for helping out!

@szymonm2, @rogerclarkmelbourne has created a variant for the [Waveshare BLE400](https://github.com/sandeepmistry/arduino-nRF5/tree/master/variants/Waveshare_BLE400). 

I2C pin info can be found here: https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/Waveshare_BLE400/variant.h#L104-L105

closed 2017-02-06 by sandeepmistry