sandeepmistry/arduino-nRF5 #92: Can't find any examples

troglodisme, 2017-02-10:
Hi there, 

I have followed all the steps on https://github.com/sandeepmistry/arduino-nRF5 but I can't seem to get anything running. 

Is there a sample example to do some simple arduino thing? Blink an led, push a button? 

My final goal is to run 8x8 display matrix over I2C using adafruit libraries. I got this working on a simple arduino uno, do you think it would run on the nRF5 dev board?

Thanks!

dlabun, 2017-02-11:
Most of the built in Arduino examples should work... Adafruit's libraries are notorious for being written specifically for the Uno and don't port over to other Arduino boards easily.

sandeepmistry, 2017-02-13:
Closing as answered.

closed 2017-02-13 by sandeepmistry