sandeepmistry/arduino-nRF5 #93: I2C pins SDA and SCL should not use S0S1 GPIO configuration.

maker-ja, 2017-02-11:
I am using this library with the sparkfun nrf52832 board. I noticed that the GPIO configuration used for I2C pins (SCA and SCL) is S0S1 (standard 0, standard 1). But open drain (S0D1 --> standard 0, disconnect 1) configuration should be used for these pins as recommended [here.](https://devzone.nordicsemi.com/question/93034/twi-clock-pull-up)

The GPIO configuration in Wire_nRF52.cpp

master = true;

NRF_GPIO->PIN_CNF[_uc_pinSCL] = ((uint32_t)GPIO_PIN_CNF_DIR_Input      << GPIO_PIN_CNF_DIR_Pos)
                                | ((uint32_t)GPIO_PIN_CNF_INPUT_Connect    << GPIO_PIN_CNF_INPUT_Pos)
                                | ((uint32_t)GPIO_PIN_CNF_PULL_Pullup      << GPIO_PIN_CNF_PULL_Pos)
                                | ((uint32_t)GPIO_PIN_CNF_DRIVE_**S0S1**       << GPIO_PIN_CNF_DRIVE_Pos)
                                | ((uint32_t)GPIO_PIN_CNF_SENSE_Disabled   << GPIO_PIN_CNF_SENSE_Pos);

NRF_GPIO->PIN_CNF[_uc_pinSDA] = ((uint32_t)GPIO_PIN_CNF_DIR_Input        << GPIO_PIN_CNF_DIR_Pos)
                                | ((uint32_t)GPIO_PIN_CNF_INPUT_Connect    << GPIO_PIN_CNF_INPUT_Pos)
                                | ((uint32_t)GPIO_PIN_CNF_PULL_Pullup      << GPIO_PIN_CNF_PULL_Pos)
                                | ((uint32_t)GPIO_PIN_CNF_DRIVE_**S0S1**       << GPIO_PIN_CNF_DRIVE_Pos)
                                | ((uint32_t)GPIO_PIN_CNF_SENSE_Disabled   << GPIO_PIN_CNF_SENSE_Pos);

I changed S0S1 --> S0D1 on my end (in the Wire_nRF52.cpp file). This should work, right?

Thanks

dlabun, 2017-02-12:
Judging from the comments from Nordic, yes that should work. Please let us know if it does work so we can update the code in the Wire library.

sandeepmistry, 2017-02-13:
@maker-ja @dlabun that looks right to me as well.

It's what the Nordic SDK has too: https://github.com/jacobrosenthal/nRF5_SDK_11.0.0/blob/78e22f2c50667c4801f84f1594f085c8b8d436a1/components/drivers_nrf/twi_master/nrf_drv_twi.c#L44-L56

@maker-ja could you please submit a pull request after testing. Thanks.

sandeepmistry, 2017-02-19:
@maker-ja any updates on this?

maker-ja, 2017-02-19:
Hi Sandeep. Sorry for delayed reply. I did test it. The I2C bus worked fine (though the S0S1 configuration also worked for my setup). I don't have a DSO to verify/ compare the authenticity of the shape of SCL waveforms in the two cases. I tested the S0D1 configuration with upto 6 slaves on the bus. I also tried with the pull-up on the uC pins disabled (as sufficient pull up was available from my sensors/ slaves). That worked fine as well. Hope this helps.  Cheers!

sandeepmistry, 2017-03-02:
Closed via #104.

closed 2017-03-02 by sandeepmistry