sandeepmistry/arduino-nRF5 #98: SoftwareSerial

andriyadi, 2017-02-22:
Is it possible to use SoftwareSerial?
I have a requirement to communicate with a serial module. While the main/hardware serial is used for debugging, need another serial port to communicate with that module.

Please advice. Thanks.

dlabun, 2017-02-24:
Maybe? I am not if anyone has tried using a Soft serial port yet.

What type of debugging are you trying to do? Just console line prints?

MartinRobotz, 2017-02-24:
Certainly not, because softserial relies on pinchange Interrupts, which itself use pin mapping. This works only on specified ports and Pins. 
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

andriyadi, 2017-02-24:
Hi @dlabun 
Yes, forgot to say that what I meant by "debugging" is "Serial.print" kind of stuff, to see what's going on. I read about RTT. Anyone has experience with it, if using RTT from Arduino environment/framework? As it seem it requires including code from Segger.

@MartinRobotz do you mean for specialized pins that supports pin interrupt, Software Serial works?
 
Thanks guys!

MartinRobotz, 2017-02-24:
For these pins it will work, but you need to know their mapping and the corresponding interrupt vector number PCINT0,1,2.... Then you have to adapt the SoftSerial library and put these interrupt vectors in the ISR. Every arduino uses different PCINT vectors for pin change interrupt. In theory that should be enough. Have a try.

dlabun, 2017-02-24:
@andriyadi What I use do to with Microchip PICs when I ran out of UARTs is run all of my debugging information out over an I2C to UART bridge. It could be an option for you, check out the NXP website below.

http://www.nxp.com/products/signal-chain/bridges/single-uart-with-i2c-bus-spi-interface-64-bytes-of-transmit-and-receive-fifos-irda-sir-built-in-support:SC16IS740_750_760#featuresExpand

sandeepmistry, 2017-02-25:
SoftwareSerial is not supported at the time.

However, I believe it should be possible to port the [AVR SoftwareSerial](https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/libraries/SoftwareSerial) and [CurieSoftwareSerial (Arduino 101)](https://github.com/01org/corelibs-arduino101/tree/master/libraries/CurieSoftwareSerial) SoftwareSerial libs to nRF5x boards. Pull requests welcome.

andriyadi, 2017-03-06:
@dlabun SC16IS740 could be one option, thanks for the idea.

Found a breakout something like [this](http://sandboxelectronics.com/?product=sc16is750-i2cspi-to-uart-bridge-module) and Arduino library like [this](https://github.com/SandboxElectronics/UART_Bridge). The library seems easy to port. Hope I can get my hands on it soon and will report the progress.

rogerclarkmelbourne, 2017-03-06:
In case its any use.

@rayburnette ported the PJRC version of SoftwareSerial  STM32 

See
http://www.stm32duino.com/viewtopic.php?f=13&t=6

(Pauls original post was here https://forum.pjrc.com/threads/17461-SoftwareSerial-doesn-t-compile-for-Teensy-3-it-seems but there may be an updated version somewhere else)

andriyadi, 2017-04-06:
I finally went to SC16IS740/50/60 route. Forked [this](https://github.com/SandboxElectronics/UART_Bridge) library and add some convenient methods. Here's my repo: https://github.com/andriyadi/UART_Bridge

I'm using [Sparkfun's breakout](https://www.sparkfun.com/products/retired/9981) to test it out.

micooke, 2017-10-09:
I am fairly familiar with the SoftwareSerial library, and this looks like a pretty up to date port : 
https://github.com/arduino-org/arduino-core-nrf52/tree/master/libraries/SoftwareSerial 

It needed a slight mod as the attachInterrupt function doesnt return the bitMask, but the problem is that it didnt work for me using this core. I tried the simple hardware/software serial echo example and it stalled after the first software print.

Very wierd as the cores are very similar, and i assume it works for the arduino core

prjh, 2017-10-09:
@micooke can confirm this! - Have done the same. but it doesen't work correctly. I swichted then over to Adafruit nRF52, because they have done the same and it does also not work (on my side)) I I asked in the Adafruit forum. I don't know but for hattach it does work - we couldn't solve it at the moment.  By the way - the original Arduino Primo SoftwareSerial on their HAL is fully functionall at all!

micooke, 2017-10-09:
@prjh - thanks for the confirmation. I might have a look at it today. The only thing i can think of it a conflict with the interrupt, which makes no sense as there isnt anything using the gpiote as far as i can tell. 

Yes, im working with the adafruit one as well - the arduino-org one seems identical

micooke, 2017-10-13:
Looking at WInterrupts.h, it looks fine. I would make a couple minor changes, but everything I've tried hasnt worked. I might try to resolve this with a simpler pinchange test first, but two three things stand out.
- GPIOTE interrupt is stupidly high priority(1), it should probably be 3 or lower
- (minor) interrupts are asynchronous, so in the Handler the for loop should break when it sees its first set interrupt and free itself up
- (minor) in attachInterrupt the configuration should be set prior to enabling the interrupt handler 

I've made these changes locally (only the first one should affect a simple example), no dice.

Having said this, I2C didn't work for me at the moment so it's possible (read likely) that my local software/hardware configuration is stuffed.

micooke, 2017-10-15:
@dlabun, @sandeepmistry - the issue is in Arduino.h, three #defines are incorrect.
``` c++
//#define digitalPinToPort(P)        ( &(NRF_GPIO[P]) ) // INCORRECT - possible seg fault
#define digitalPinToPort(P)        ( NRF_GPIO )
//#define portOutputRegister(port)   ( &(port->OUTSET) ) // INCORRECT
#define portOutputRegister(port)   ( &(port->OUT) )
//#define portModeRegister(port)     ( &(port->DIRSET) ) // INCORRECT
#define portModeRegister(port)     ( &(port->DIR) )
```
I can create a PR with this fix. Are are you interested in supporting the SoftwareSerial library as i can pull this in at the same time?

With these fixes (@prjh - this is the same problem in the adafruit core, they have 2 incorrect), the SoftwareSerial library from Adafruit (https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/SoftwareSerial) or Arduino (https://github.com/arduino-org/arduino-core-nrf52/tree/master/libraries/SoftwareSerial) should work fine. My local version is based off the adafruit one (the arduino one looks identical) hacked with Tx/Rx only & half-duplex options, so it will take a while to regression test these for full confirmation.

prjh, 2017-10-16:
@micooke Congratulation! I have tested SoftwareSerial with the attachInterrupt() changes and the above #defines well with a nRF52-DK board. Up to 38400 baud, it does work straight forward. In my tests with the Arduino Primo implementation I got baud rate up to 74880 - don't know why at the moment. Never the less it does work - Thanks for your investigation.

micooke, 2017-10-17:
I have created a PR for the core side without a SoftwareSerial library, but i am happy to add this in.
See PR #205

micooke, 2017-10-25:
Just a note that ive added  SoftwareSerial to the PR.

@prjh - im not sure why the primo out performs?

I will say that the tuning is very coarse for most non-avr variants of this library as they use ```delayMicroseconds()```, whereas the avr uses  ```_delay_loop_2()``` from ```<util/delay_basic.h>``` which is sub-microsecond (4 clock cycles). In other words, this library won't work at high speeds as the tuning will be off - which is what you have observed.

idreamsi, 2018-04-09:
a question:
Is **SoftwareSerial** currently usable?

micooke, 2018-04-09:
@idreamsi - you can always try the PR associated with this, #205. This is still waiting a merge

idreamsi, 2018-04-09:
@micooke - I tested it, It works. Thank you very much ;)

micooke, 2018-04-09:
You are very welcome

shajek, 2019-03-05:
any update or plan for out of box software serial? or some complete tutorial how do get it work now. Thank you (if not, that please write some verified I2C - UART bridge)