sandeepmistry/arduino-nRF5 #99: Erasing a SoftDevice

TTrml, 2017-02-24:
Hello,

Thanks for this awesome repo (and the many others)!

Is it possible to erase the softdevice on a chip, or even erase its full content?
I tried to do a "burn bootloader" with the Softdevice set to "None", but it fails. That would be a nice feature, since it would obviate the need for nRF Go Studio.
Best regards,

Armel

dlabun, 2017-02-24:
There's no button in the IDE to nuke a chip back to a clean slate, but you can certainly do it manually. I'll try to help you out with instructions... What board and OS are you using?

TTrml, 2017-02-24:
Thanks! I am using Windows, with an ST-Link V2 to program nRF51/nRF52-based custom boards. Works like a charm with this core.

sandeepmistry, 2017-02-25:
@TTrml good suggestion, I think this might be possible with the master version, when the soft device is flashed the entire flash is erased first. (This will be available in the next release as well)

bossesand, 2017-03-10:
Just a question, What version of S132 is the .h files for in the current release of arduino-nRF5 ?
The Latest S132.5.0.0.1-alfa does not seem to be backwards compatible.

sandeepmistry, 2017-03-13:
@bossesand

> What version of S132 is the .h files for in the current release of arduino-nRF5 ?

 s132_nrf52_2.0.0

kriswiner, 2017-04-05:
You can use nRFgo Studio and erase the chip.

d00616, 2017-04-25:
Hello,

with openocd, there is an mass_erase option: https://github.com/RIOT-OS/RIOT/wiki/Board%3A-yunjia-nrf51822#protected-at-factory-error-writing-to-flash-at-address-0x000-

In my EEPROM emulation library, there is an example sketch "flash_erase_all". The library is available via library manager as "arduino-NVM" or at https://github.com/d00616/arduino-NVM

micooke, 2017-08-21:
Just a thought on this, the flash is erased prior to uploading the soft device - will openocd allow you to upload an empty hex?

Otherwise openocd would have to support a conditional statement and i dont see why openocd would support this.

Actually no, it would need to be through a bat file or script or jar if uploading an empty hex doesn't work.

micooke, 2017-08-22:
I just checked this. An error is thrown if you select 'None', however as the erase occurs prior to trying to upload a hex (that doesn't exist) the target is erased.

Ive made some changes locally to stop the error, if there is interest i can create a pull request. Basically i created an 'empty' hex (object header and footer but no contents), created the appropriate directory structure and added a none.softdeviceversion= to all variants in boards.txt. 

It's easy to fix but is it really worth it considering this patch is purely to stop Arduino reporting that an error occured when trying to upload a 'none' or null bootloader?

sandeepmistry, 2017-09-05:
@micooke nice a pull request would be awesome. In general, anything with a "help wanted" tag is likely to eventually get merged if a PR is available :)

micooke, 2017-09-05:
Will do. I stupidly commited the bmp changes to the master of my fork, so once ive added the recommended changes ill send a PR to resolve this.

sandeepmistry, 2017-09-11:
Fixed via @micooke's https://github.com/sandeepmistry/arduino-nRF5/pull/191/commits/0a3b66e52494f78daccde71349e810fe7174ff32.

With the latest Github master version, soft devices can be erased by:
1) Selecting "None" from the Tools -> SoftDevice menu
2) Then selecting "Burn Bootloader" from the Tools menu.

closed 2017-09-11 by sandeepmistry