sandeepmistry/arduino-nRF5 #101: Linux Arduino runtime.tools path errors

erickrupinski, 2017-02-25:
Arduino 1.8.1 installed from website, nRF 0.2.0, Xubuntu 16.04

Strings with runtime.tools do not appear to be properly pathed or permissioned? 

I hardcoded arm-none-eabi-g++ to the system arm-none-eabi-g++ version of the file by changing the platform.txt. I was able to get past the first error I encountered, and then encountered another path error with the openocd file. 

#HackIllinois2017

```
/home/eric/Downloads/arduino-1.8.1/arduino-builder -dump-prefs -logger=machine -hardware /home/eric/Downloads/arduino-1.8.1/hardware -hardware /home/eric/.arduino15/packages -tools /home/eric/Downloads/arduino-1.8.1/tools-builder -tools /home/eric/Downloads/arduino-1.8.1/hardware/tools/avr -tools /home/eric/.arduino15/packages -built-in-libraries /home/eric/Downloads/arduino-1.8.1/libraries -libraries /home/eric/Arduino/libraries -fqbn=sandeepmistry:nRF5:Generic_nRF51822:chip=xxaa,softdevice=none,lfclk=lfxo -ide-version=10801 -build-path /tmp/arduino_build_370034 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=/home/eric/.arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5 -prefs=runtime.tools.gcc-arm-none-eabi.path=/home/eric/.arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4 -verbose /home/eric/Downloads/arduino-1.8.1/examples/01.Basics/Blink/Blink.ino
/home/eric/Downloads/arduino-1.8.1/arduino-builder -compile -logger=machine -hardware /home/eric/Downloads/arduino-1.8.1/hardware -hardware /home/eric/.arduino15/packages -tools /home/eric/Downloads/arduino-1.8.1/tools-builder -tools /home/eric/Downloads/arduino-1.8.1/hardware/tools/avr -tools /home/eric/.arduino15/packages -built-in-libraries /home/eric/Downloads/arduino-1.8.1/libraries -libraries /home/eric/Arduino/libraries -fqbn=sandeepmistry:nRF5:Generic_nRF51822:chip=xxaa,softdevice=none,lfclk=lfxo -ide-version=10801 -build-path /tmp/arduino_build_370034 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=/home/eric/.arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5 -prefs=runtime.tools.gcc-arm-none-eabi.path=/home/eric/.arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4 -verbose /home/eric/Downloads/arduino-1.8.1/examples/01.Basics/Blink/Blink.ino
Using board 'Generic_nRF51822' from platform in folder: /home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0
Using core 'nRF5' from platform in folder: /home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0
Detecting libraries used...
"/home/eric/.arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0 -mthumb -c -g -Os -w  -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10801 -DARDUINO_GENERIC -DARDUINO_ARCH_NRF5  -DNRF51 -DNRF5 "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/toolchain/" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/toolchain/CMSIS/Include" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/toolchain/gcc/" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/device/" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/drivers_nrf/delay/" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/softdevice/none/headers/" -DUSE_LFXO "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/variants/Generic" "/tmp/arduino_build_370034/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/home/eric/.arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0 -mthumb -c -g -Os -w  -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10801 -DARDUINO_GENERIC -DARDUINO_ARCH_NRF5  -DNRF51 -DNRF5 "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/toolchain/" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/toolchain/CMSIS/Include" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/toolchain/gcc/" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/device/" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/drivers_nrf/delay/" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/softdevice/none/headers/" -DUSE_LFXO "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5" "-I/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/variants/Generic" "/tmp/arduino_build_370034/sketch/Blink.ino.cpp" -o "/tmp/arduino_build_370034/preproc/ctags_target_for_gcc_minus_e.cpp"
fork/exec /home/eric/.arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-g++: no such file or directory
Error compiling for board Generic nRF51822.
```

```
/usr/bin/arm-none-eabi-gcc"  "-L/tmp/arduino_build_370034" -Os -Wl,--gc-sections -save-temps  "-L/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/toolchain/gcc/" "-L/home/eric/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/softdevice/none/toolchain/armgcc/" "-Tnrf51_xxaa.ld" "-Wl,-Map,/tmp/arduino_build_370034/Blink.ino.map" --specs=nano.specs --specs=nosys.specs -mcpu=cortex-m0 -mthumb  -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -o "/tmp/arduino_build_370034/Blink.ino.elf" "/tmp/arduino_build_370034/sketch/Blink.ino.cpp.o" "/tmp/arduino_build_370034/core/variant.cpp.o" -Wl,--start-group -lm "/tmp/arduino_build_370034/core/core.a" -Wl,--end-group
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .heap by 4 bytes
"/usr/bin/arm-none-eabi-objcopy" -O binary  "/tmp/arduino_build_370034/Blink.ino.elf" "/tmp/arduino_build_370034/Blink.ino.bin"/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .stack_dummy by 4 bytes

/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .heap by 4 bytes
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .stack_dummy by 4 bytes
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .heap by 4 bytes
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .stack_dummy by 4 bytes
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .heap by 4 bytes
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .stack_dummy by 4 bytes
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .heap by 4 bytes
/usr/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: warning: changing start of section .stack_dummy by 4 bytes
"/usr/bin/arm-none-eabi-objcopy" -O ihex  "/tmp/arduino_build_370034/Blink.ino.elf" "/tmp/arduino_build_370034/Blink.ino.hex"
Sketch uses 1496 bytes (0%) of program storage space. Maximum is 262144 bytes.
java.io.IOException: Cannot run program "/home/eric/.arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5/bin/openocd": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingProgrammer(SerialUploader.java:343)
[... cut ...]
	at processing.app.SketchController.exportApplet(SketchController.java:682)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2190)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch
```

dlabun, 2017-02-25:
I don't use Linux at all but from the error message it looks like OpenOCD isn't even installed... Can you verify if it there?

sandeepmistry, 2017-02-25:
The toolchains are currently 32-bit on Linux, so for Travis CI `libc6:i386` and `libstdc++6:i386` need to be installed, something to PR to the read me?

erickrupinski, 2017-02-26:
Good catch. I was running 64 bit linux when I first ran the toolchains. I tried running it again in a 32 bit installation of Xubuntu and did not run across these errors.

jacobrosenthal, 2017-04-10:
@tlonch Added some linux install instructions in #143

closed 2017-04-10 by jacobrosenthal