sandeepmistry/arduino-nRF5 #102: Tiny BLE soft device errors

tsong08, 2017-02-25:
A common error we keep seeing is:

```
fatal error: ble_gatts.h: No such file or directory

compilation terminated.```

If you see this error, the solution is to go to tools>softdevices>select softdevice.

Include in FAQ for future reference?

#HackIllinois2017

dlabun, 2017-02-25:
Thanks for the suggestion. I assume this is occurring when using the BLE Peripheral library?

sandeepmistry, 2017-02-25:
@tsong08 yes please, PR to the read me would be great.

tsong08, 2017-02-25:
@dlabun Yes this is specific to the BLE Peripheral code. Basic sketches compile fine.

dlabun, 2017-02-25:
@sandeepmistry Should the note be here, on the BLE-Peripheral library or both?

sandeepmistry, 2017-02-25:
There's a note here: https://github.com/sandeepmistry/arduino-nRF5#ble

However, maybe it needs to be clearer. We can also update the BLEPeripheral read me.

dlabun, 2017-02-25:
Maybe something along the lines of "trying to use the BLEPeripherial library without selecting a Soft Device in the menu will result in build errors"?

jacobrosenthal, 2017-02-26:
Yeah I think its just a 'bad' error message, presuming we cant detect and
generate a better one instead, itd be good to document specifically that
this error should be translated as 'please select a softdevice'
```

fatal error: ble_gatts.h: No such file or directory

```

sandeepmistry, 2017-02-27:
@jacobrosenthal @dlabun those sound like good ideas, any ones have time to prep a PR?

jacobrosenthal, 2017-03-02:
Can close this for https://github.com/sandeepmistry/arduino-BLEPeripheral/pull/129

closed 2017-04-10 by jacobrosenthal