sandeepmistry/arduino-nRF5 #106: Switch to mainline openocd

sandeepmistry, 2017-03-02:
Now that the following patch has been integrated into the mainline OpenOCD: http://openocd.zylin.com/#/c/3510/

I think we can switch to the GNU ARM Eclipse version of OpenOCD. We just need to repackage the archives into a format the Arduino IDE expects:
https://github.com/gnuarmeclipse/openocd/releases

As well as modify the `tcl/target/nrf52.cfg` file or use a local config version stored in the repo: http://openocd.zylin.com/#/c/3399/

sandeepmistry, 2017-03-02:
This also will give us 32-bit and 64-bit versions of the Linux binary.

carlosperate, 2017-03-27:
Could ARM Linux be supported out of the box as well with this?

sandeepmistry, 2017-04-10:
@carlosperate If something provides a build of the toolchains for ARM Linux, I'd be happy to add it to the JSON :)

carlosperate, 2017-04-10:
Well, if we could use upstream openOCD I was hoping to be able to use pre-build binaries from teams like Eclipse: https://github.com/gnuarmeclipse/openocd/releases
But I am not sure how often do they update their binaries.

sandeepmistry, 2017-04-10:
I don't see and ARM openocd binary there, we also need an ARM to ARM compiler?

carlosperate, 2017-04-10:
Yeah, sorry, I edited my previous message once I realised they didn't build for ARM.
We would need ARM GCC, but if I was to try this myself I would just build it in a ARMv6 raspberry pi with an old version of Raspbian (and therefore an old version of glibc), which should be fine for most debian-based ARM distributions (and hopefully other distros as well, but hard to tell).

dlabun, 2018-02-09:
@sandeepmistry Is there still a desire to move to the mainline OpenOCD? While I am working on the repository this weekend I can spend some time repackaging OpenOCD as well.

sandeepmistry, 2018-02-10:
@dlabun yes, that would be great. We might have to [patch](https://github.com/sandeepmistry/openocd-code-nrf5/commit/26ca830915275015be243a3fcc074856621e6d08) `tcl/target/nrf52.cfg` from the mainline OpenOCD version. I suggest we repackage the "GNU ARM Eclipse version of OpenOCD" to a archive format that is compatible with the Arduino IDE, instead of rebuilding it from source.