sandeepmistry/arduino-nRF5 #107: Add mbed disk programmer option

sandeepmistry, 2017-03-02:
During #HackIllinois2017 @jacobrosenthal observed many issues with Windows and Zadig.

Need to explore adding a option to upload using the mbed disk using a batch file (windows) and shell script (Linux/Mac).

Rough set of steps needed:
1) Build .hex file and merge soft device. If soft device is not erased during upload we can skip the merge step.
2) Detect mbed drive location via checking drives for a MBED.htm file
3) Copy file to mbed disk drive.

dlabun, 2017-03-02:
I've been using the Segger upload tools instead of the current method and I find it's more reliable. All you need is the Nordic tool kit (free) and a batch file on Windows. There's no code to maintain other than the batch file. There's also the benefit of supporting normal, non-integrated Segger debug probes.

sandeepmistry, 2017-03-02:
@dlabun could this approach work for Mac and Linux too?

@jacobrosenthal what do you think of this? I think the one advantage of the mbed approach is it would also work with CMSIS DAP programmers too.

dlabun, 2017-03-02:
Yes it would as Nordic has the tool kit for Mac and Linux as well.... They call it the Command Line Tools on the link below.

http://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52-DK#Downloads

dlabun, 2017-03-02:
As a side note, Nordic has updated the OB-JLink firmware to rev 160914.

jacobrosenthal, 2017-03-02:
> nRF52-DK#Downloads

Id say go with the segger stuff. The mbed disk is pretty pervasive, but
theres no feedback except the fail file and we dont want to be digging
around in there for success/fail messages

That said, license issues? Id hate to add a fourth component to download
its already pretty arduous. I guess it could be bundled with the softdevice
download?

dlabun, 2017-03-02:
As far as I can tell there shouldn't be licensing issues as the tool kit is freely available on the Nordic site, it's being used strictly on Nordic chips as intended and we wouldn't be modifying it.  

Adding a fourth component is a good point... Is there a way to make it an optional part of the install? (If you want it, go grab it?)

jacobrosenthal, 2017-03-02:
Wait, what sandeep said.

The segger stuff will only work with jlink devices, and the mbed stuff is
going to be cmsisdap based..

dlabun, 2017-03-02:
Hmm... You almost need 3 different upload methods to cover all of the tools out there. May you could do 3 different editions of the core? For example v0.3-OpenOCD, -Segger or -mbed... Only difference being the flashing tool that installed.

jacobrosenthal, 2017-03-02:
I dont love that. If theres no licensing then just bundle the smallest
subset and make it a larger download.

dlabun, 2017-03-02:
Well the Linux package was 770kb to download, Win32 installer was 24mb... I don't think that will be too bad

jacobrosenthal, 2017-03-02:
I wonder if this is just a virtual serial port driver, or if it does
anything with cmsisdap. Not sure we tried it this weekend.
https://developer.mbed.org/handbook/Windows-serial-configuration#1-download-the-mbed-windows-serial-port

sandeepmistry, 2017-03-02:
The DAP part is based on USB HID, not sure if you need a driver for it.

Should I rename this issue to something more like "Improve Windows setup"?

jacobrosenthal, 2017-03-02:
Ill get a windows vm and see what the hell it shows up as

On Wed, Mar 1, 2017 at 7:17 PM, Sandeep Mistry <notifications@github.com>
wrote:

> The DAP part is based on USB HID, not sure if you need a driver for it.
>
> Should I rename this issue to something more like "Improve Windows setup"?
>

dlabun, 2017-03-02:
So I just this very bolded warning on the mbed website regarding the nrf51 and 52 DKs... Could be rather inconvenient

Please note that there is a behaviour change between this firmware and the J-Link firmware - with the mbed firmware you must press the 'BOOT/RESET' button the board to run your code once it has been downloaded *

jacobrosenthal, 2017-03-02:
So I figured out drivers mostly. I think we could use the mbed as an option and if it works for most people so be it.

tested in windows 8 vm

For a tinyble mbed device:
seems like this mbed driver link installs serial and hid drivers!
https://developer.mbed.org/handbook/Windows-serial-configuration#1-download-the-mbed-windows-serial-port

For my nrf51 dongle (segger jlink), after windows takes FOREVER to install its driver:
zadig instructions are correct, but interface 2 is called BULK

jacobrosenthal, 2017-03-02:
Also I think maybe better roll back instructions are from https://github.com/pbatard/libwdi/issues/8
Device manager -> right click on device -> uninstall and make sure to check box for "Delete the driver software for this device"

jacobrosenthal, 2017-03-02:
@dlabun at least for tinyble and nrf51 dongle theres no manual reset needed *shrug*

sandeepmistry, 2017-03-03:
> Also I think maybe better roll back instructions are from pbatard/libwdi#8

For sure, there's a similar PR for `node-bluetooth-hci-socket`: https://github.com/sandeepmistry/node-bluetooth-hci-socket/pull/54#issuecomment-277601713

A nice gif would be sweet as well.

@dlabun does your JLINK also have a USB MSD disk? The one in my nRF52DK does, not sure if this means I'm using the mbed f/w.

dlabun, 2017-03-03:
@sandeepmistry Yes, there's a USB MSD with the jlink firmware.

sandeepmistry, 2017-03-13:
Ok, then I think we can just copy of the hex file mbed style with the jlink firmware too.