sandeepmistry/arduino-nRF5 #108: Wire library

kriswiner, 2017-03-02:
I am using my own custom nRF52832 development boards and I would like to initialize the I2C bus, which I have placed on pins 6 (SDA) and 7 (SCL) out of convenience. I looked in the nRF52 DK variant files and tried to change the pin numbers there but it is pretty confusing.  I managed to figure out the pin labelling for the leds of the board but how can I enable I2C on pins 6 and 7. Is the default on pins 26 and 27 or pins 20 and 21, and are these pins the P0.20 and P0.21 pins or physical pins 20, 21 or what?

Wouldn't it make more sense to have a general nRF52 variant and let the user choose as in:

Serial.begin(RX_pin, TX_pin, baudrate);  and

Wire.begin(SDA_pin, SCL_pin); ?

Also, I tried to use TimerOne.h from the BLEPeripheral library examples and the IDE complains that Timer1.initialize and Timer1.attachinterrupt are not a Timer1 class.

Is there no other way to use timers for the nRF52?

Thank you for our help!

kriswiner, 2017-03-05:
I was able to change the SDA/SCL pins in the variant.cpp to the correxct pins 6 and 5 in my case but I cannot get an ACK on the I2C Scanner. The SCL line is being held down tp 0.11 V. Not quite sure I understand this behavior. I tried on another pair of pins and I see the same behavior. Something in the Wire implementation is holding the SCL line. 

Is there any example of writing to a sensor and reading its data?

kriswiner, 2017-03-05:
I am getting error = 2 or NACK on all addresses up to 0x67 and then on 0x68, which is the address of the MPU9250, an I2C device on the bus, I am getting nothing from endTransmission, the bus is hanging. Do you know what the reason is for this?

dlabun, 2017-03-05:
What value are your I2C pull up resistors?

kriswiner, 2017-03-05:
4K7, this board works fone with mbed, it is not a hardware issue.

dlabun, 2017-03-05:
What version of the core are you using?

kriswiner, 2017-03-05:
0.2.0

Is there a newer or better one?

dlabun, 2017-03-05:
v0.3 is out now, and we did make a change to the configuration of the I2C bus. You may want to try it and see if there is any difference.

kriswiner, 2017-03-05:
updating now...

dlabun, 2017-03-06:
Have you tried any of the libraries for the MPU9250 ? I can't judge if your scanner code is working correctly or not.

kriswiner, 2017-03-06:
The scanner code simply sends a transmission to all I2C devices on the bus
and reports the error condition. The hardware works fine using mbed, as
does the scanner with any MCU.

Do you have a scan example that returns a proper ACK from any I2C device on
the nRF52 bus? Do you have a working example of successful I2C transations
using the nRF52 with any I2C device?

There is something wrong with the nRF52 Wire.h library IMHO...

dlabun, 2017-03-06:
I do not have a scan example as I use an oscilloscope to verify bus activity. I am reluctant to condemn the library itself as this is the first report of an issue and so far I haven't seen issues with the Nordic development kits. For the sake of troubleshooting, I'd recommend trying a library for the MPU9250 or swap it out for a different I2C device.

kriswiner, 2017-03-06:
I have tried this on two different pcbs, and there is a BMP280 on the i2C
bus as well, and I have added a different I2C deviceon a breadboard. They
all behave the same way. No ACK from any I2C device and the first I2C
device to be polled holds the SCL ine LOW. Not sure what in the software is
doing this, but it is not a hardware issue.

dlabun, 2017-03-06:
My only other suggestion would be to try one of the supported boards to see if the issue can be duplicated. Other than that, you will need to wait until Sandeep can look into this.

kriswiner, 2017-03-06:
OK, had to force use of the nRF51 Wire.cpp file and now everything works, although I am getting some pretty strange data values, at least I am getting proper ACK from the I2C devices. Something definitely wrong with the nRF52 Wire.cpp; I would compare with what you're using for the nRF51...

closed 2017-03-06 by kriswiner

bossesand, 2017-03-10:
Is the driving current for the i2c bus configured for high current?
As I have read if the pullups are below 10kohm you need to configure for high current and may also disable the internal pullups.
How to do this in the Arduino setup I do not how to do this.

kriswiner, 2017-03-10:
I switched to the nRF51Wire.h and everything works. I use 4K7 pullups and
do not do anything to configure the I2C bus for high current.

bossesand, 2017-03-10:
Interesting, what is the difference?

kriswiner, 2017-03-10:
52 is non-blocking with DMA and 51 is blocking w/o DMA, but otherwise I
don't know why one works and one doesn't.

bossesand, 2017-03-12:
I am quite a beginner in programing 
I have this board  Taida_Century_nRF52_minidev\variant.cpp 

I use a Salea 8CH to watch pins P02 and P03 no pinchanges only from 0 to Vcc when powering on.

I have choosen the Taidain the GUI, and it compiles for cortex-m4, but  when I compile I get this warning, is it relevant?

Using board 'STCT_nRF52_minidev' from platform in folder: C:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0
Using core 'nRF5' from platform in folder: C:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0
Warning: Board arduino:avr:Generic_nRF51822 doesn't define a 'build.board' preference. Auto-set to: AVR_GENERIC_NRF51822
Detecting libraries used...
"C:\Users\bosse\AppData\Local\Arduino15\packages\SparkFun\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10801 -DARDUINO_STCT_NRF52_minidev -DARDUINO_ARCH_NRF5  -DNRF52 -DS132 -DNRF51_S132 -DNRF5 "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/CMSIS/Include" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/gcc/" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/device/" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/drivers_nrf/delay/" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/softdevice/s132/headers/" -DUSE_LFXO "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\variants\Taida_Century_nRF52_minidev" "C:\Users\bosse\AppData\Local\Temp\arduino_build_686306\sketch\Blinky-ble400-bosse.ino.cpp" -o "nul"
"C:\Users\bosse\AppData\Local\Arduino15\packages\SparkFun\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10801 -DARDUINO_STCT_NRF52_minidev -DARDUINO_ARCH_NRF5  -DNRF52 -DS132 -DNRF51_S132 -DNRF5 "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/CMSIS/Include" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/gcc/" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/device/" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/drivers_nrf/delay/" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/softdevice/s132/headers/" -DUSE_LFXO "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5" "-IC:\Users\bosse\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\variants\Taida_Century_nRF52_minidev" "C:\Users\bosse\AppData\Local\Temp\arduino_build_686306\sketch\Blinky-ble400-bosse.ino.cpp" -o "C:\Users\bosse\AppData\Local\Temp\arduino_build_686306\preproc\ctags_target_for_gcc_minus_e.cpp"

If I try to compile blinky version modified from BLE400 I get this error
C:\Users\bosse\Documents\Arduino\Blinky-ble400-bosse\Blinky-ble400-bosse.ino:28:22: fatal error: nrf_gpio.h: No such file or directory

Detecting libraries used as kriswiner suggested
I have tried to change the wire,h and wire.ccp from nrf%1 to nrF52 version.

in variant.h -- Does this define PO02 and PO03 as ourputs for i2c?
#define PIN_WIRE_SDA         (2u)
#define PIN_WIRE_SCL         (3u)

But I see no change at all on pin Po02 and po03
in Wire.h - should this be removed? - probaly not....
#ifdef NRF52
    TwoWire(NRF_TWIM_Type * p_twim, NRF_TWIS_Type * p_twis, IRQn_Type IRQn, uint8_t pinSDA, uint8_t pinSCL);

I see no indications on any of the PO00 - PO10 pins showing 12c activity

Kris could you send me a compiled version of your i2c scanner? on which pins is the i2c for you?
Bo-Erik

bossesand, 2017-03-12:
A bit of black magic.
Changed around my initialization to be;
void setup(){
    Wire.begin();
    Serial.begin(115200);
    mySensor.init();
}
instead of calling sensor initilization before wire.begin now the sda scl pins are alive :-)
I used the master_reader example in wire to verify that the i2c bus was alive...

sandeepmistry, 2017-03-13:
@bossesand please open a new issue to track what you are seeing, this one has been closed by the author and might be unrelated.

reapqin, 2017-04-27:
@kriswiner i am working with nrf52832 + mpu9250 also. And i see your commet about the "I switched to the nRF51Wire.h and everything works"
could you tell me how to switch that? I am begginer wish your help.

i am using a genneric nrf52832 [https://item.taobao.com/item.htm?spm=a230r.1.14.74.IuClmR&id=532630623264&ns=1&abbucket=15#detail](url) with all pins out around borad. and i changed the sda to 26u scl to 27u in variant.h in genneric fold
and i found the board pins 26 27 was not pullup with resistors. is it nessory?
Wish your help thank you.

kriswiner, 2017-04-27:
> htm?spm=a230r.1.14.74.IuClmR&id=532630623264&ns=1&abbucket=15#detail

You can open an issue on the nRF52 arduino site since they claim I2C works
fine.

mohandurga, 2017-05-31:
hello,
I have a same problem with I2C as you discussed in the above conversions @kriswiner 

I am getting error = 2 or NACK on all addresses up to 0x67 and then on 0x68, which is the address of the MPU9250, an I2C device on the bus, I am getting nothing from endTransmission, the bus is hanging. Do you know what the reason is for this?

@kriswinerr can you please guide me to  solve this problem
Thank you

bossesand, 2017-05-31:
I have not had this problem recently with my bno055, I am even running it
at 400kHz now.
If I remember it correctly I beleive my problem was that I had switched the
read/write and data cables.
I bit embarassing but I think that was the problem.

A low cost logic analyzer from Aliexpress assisted me greatly in finding
this. :-)

BR Bo-Erik

VRTracker, 2017-07-17:
I am experiencing the same problem as described by @kriswinerr and @mohandurga. 
Using a NRF Thingy programmed using and NRF 52840 DK.
It always hangs on EndTransmission. Using Wire_nRF51 instead of Wire_nRF52 didn't help for me.

Does anybody understand what could be going on ?
Thank you !

rogerclarkmelbourne, 2017-07-17:
Did you connect a logic analyser ?

BTW. Isn't the Thingy a nRF52 device so you will need to make sure you are using the nRF52 version of the Wire library

VRTracker, 2017-07-18:
Ok my bad, really dumb problem : the pinout was not right. With nRF Thingy 52 the pins on the PCB called "SDA and SCL" are actually "SDA_EXT SCL_EXT" (pins 14-15).

I saw your problem @rogerclarkmelbourne, I have not experienced problems with that so far but if I do I'll tell you about it.

rogerclarkmelbourne, 2017-07-18:
Thanks.  The proble with extra clocks does not seem to be a problem with most I2C slaves, but it was with the one that I am using

gtalusan, 2017-10-20:
Having a similar problem with 0.4.0 and a RedBear BLE Nano 2.  The sketch will freeze at Wire.endTransmission.

I switched over to Wire_nRF51 and changed the #ifdef's accordingly and the typical I2C Scan sketch worked.