sandeepmistry/arduino-nRF5 #110: tools are very difficult to populate when not using board manager

jacobrosenthal, 2017-03-02:
the tooling (openocd and arm gcc) currently download into top level directory .../Arduino15/packages/sandeepmistry/tools 

This is good because it means when the core is updated the tooling is only updated if it needs to be resulting in a fair bit less downloading per release.

However, this makes it difficult to install the core without a board manager or to develop a toolchain update because in platform.txt we hardcode that path with ```runtime.tools.openocd-0.10.0-dev.nrf5.path``` which ONLY resolves to the .../Arduino15/packages/sandeepmistry/tools folder and not the sketchbook folder

This means to populate tools you need to
* manually download toolchains to .../Arduino15/packages/sandeepmistry/tools or
* dig in and patch the path in the platform.txt
Both of which break the whole point of having the hardware directory be available in the sketchbook without digging around system folders

Ive been chatting with Sandeep about this and he thinks theres a way to set platform.txt to patch platform.txt as part of the release process, to utilize the 

That would also require changing the way pathing currently works to find the tools in something like ../{runtime.platform.path}

Or also we could just document this mess in the current readme as I find the current verbiage misleading as git cloning the core doesnt give you a working board. Thoughts?

dlabun, 2017-03-02:
Has this question been raised up to the Arduino folks? I see this being more of a global question other 3rd party core developers may want answered.

sandeepmistry, 2017-03-03:
> Or also we could just document this mess in the current readme as I find the current verbiage misleading as git cloning the core doesnt give you a working board. Thoughts?

So is the first step unclear do we think a link to the anchor? https://github.com/sandeepmistry/arduino-nRF5#from-git-for-core-development
> 1. Follow steps from Board Manager section above

We definitely need to change it to say this pulls down the toolchain.