sandeepmistry/arduino-nRF5 #111: Document programmer to use with standard boards

jacobrosenthal, 2017-03-02:
Its currently a guessing game to know that the nrf51 dongle uses a jlink programmer, but mbed boards use a cmsis dap, etc, etc.

It would be good at the very least to compile a list of these. Anyone want to drop a comment which programmer to use with their favorite boards?

Then where to put that, we list the supported boards in the readme. .It could go there. This is clumsy, but we could also try to smoosh it into the board name in the arduino menu item?

jacobrosenthal, 2017-03-02:
I see a upload.interface defined as jlink and cmsis-dap in boards.txt but not sure that it does anything. Anyone know what this is for or a link for how arduino uses it? Can it select the appropriate programmer by default?

sandeepmistry, 2017-03-03:
> Anyone want to drop a comment which programmer to use with their favorite boards?

Most of the boards I use have built-in CMSIS DAP or JLINKS, I use a STLink clone with my BluzDK.

> Anyone know what this is for or a link for how arduino uses it? Can it select the appropriate programmer by default?

See [`tools.openocd.upload.pattern="{path}/{cmd}" {upload.verbose} -f interface/{upload.interface}.cfg -c "{upload.setup_command}" -f target/{upload.target}.cfg -c "program {{{build.path}/{build.project_name}.hex}} verify reset; shutdown;"`](https://github.com/sandeepmistry/arduino-nRF5/blob/master/platform.txt#L119)

Here's my understanding, if the boards.txt entry specifies `<board>.upload.protocol=` the `tools.openocd.upload.pattern` is used, other wise `tools.openocd.program.pattern` is used. You can also select `Sketch -> Upload Using Programmer` for force using the programmer.

jacobrosenthal, 2017-03-03:
Super Interesting. None of the boards use upload.protocol. they do have upload.interface, and a bunch of them are defined as cmsisdap, which if it was supposed to be fed into openocd would be wrong (cmsis-dap.cfg is the setting)

Any idea if upload.interface is valid/used? 

I also needed 
TinyBLE.upload.setup_command=transport select swd;

dlabun, 2017-03-03:
All of my stuff is programmed with a jlink onboard chip or my external jlink

jacobrosenthal, 2017-03-04:
Also of note, softdevice upload still requires a programmer to be set sadly. It seems like this could/should be altered to use the program pattern to be overridden from boards.txt

sandeepmistry, 2017-03-13:
@jacobrosenthal I'm sure the Arduino IDE team would be up for something like this ;)

Best way to proceed is to open a PR with the proposal.

sandeepmistry, 2017-03-26:
@jacobrosenthal what do we want to do with this?

jacobrosenthal, 2017-03-27:
I think this mostly goes away with #114 but would be nice to stick around to track arduino upstream support for bootloader overrides, whenever I get around to trying to PR that

jacobrosenthal, 2017-04-10:
This is mostly solved now that #114 is merged

closed 2017-04-10 by jacobrosenthal