sandeepmistry/arduino-nRF5 #116: Is there support for I2C in the library?

XaviLC, 2017-03-06:
Hello,

I was wondering whether or not I can make use of the I2C (TWI in Nordic code) with the current library or not. I have attached some sensor to the sdla/sci pins (plus vcc and gound)

jacobrosenthal, 2017-03-06:
Yep. Ive seen working code on nrf51 boards. I think theres an open issue
with i2c on the nrf52 though..
https://github.com/sandeepmistry/arduino-nRF5/issues/108

On Mon, Mar 6, 2017 at 9:52 AM, XaviLC <notifications@github.com> wrote:

> or not I can make use of the I2C (TWI in Nordic code) with the current
> library or not. I have attached some sensor to the sdla/sci pins (plus vcc
> and gound)
>
>

XaviLC, 2017-03-06:
Thanks Jacob,

But in that topic they are trying to change the I2C pins whereas I would like to keep them as they are. Would I still need to implement the scanner ?

kriswiner, 2017-03-06:
I cannot get I2C to work on the nRF52 with any pin pair. The SCL line is held by the I2C device and does ot return an ACK. This seems to be a software problem. I just switched pins pairs to verify this.

XaviLC, 2017-03-06:
How do you mean a software problem? That the configured voltage of the line is too low?

dlabun, 2017-03-06:
@XaviLC I2C does work on the boards listed in the ReadMe file. There may be an issue with custom / unsupported boards and we are trying to figure out that issue.

XaviLC, 2017-03-06:
@dlabun, that would mean that since my board is listed in the readme, I shouldn't have that issue?

dlabun, 2017-03-06:
Correct, you shouldn't have an issue... But if run into an issue please let us know.

XaviLC, 2017-03-06:
@dlabun I have connected my Adafruit_TSL2591 sensor(I am making using as well of their lib) to pins 26 and 27 (sda and scl) and I didn´t get anything from the sensor, I will check voltage level just incase

dlabun, 2017-03-06:
Which board are you using?

XaviLC, 2017-03-06:
I am using the pca 10040 s132

dlabun, 2017-03-06:
I wouldn't expect an issue with the nRF52 Development kit and an Adafruit breakout board. I have an Adafruit breakout for an accelerometer, let me test it and see if there's something going on.

nihaopaul, 2017-03-07:
i've had no problem with i2c setup and usage, only thing i had to do was
swap the two pins around for my custom board and remember to terminate with
the correct cap/resistors to prevent it from freezing (datasheet of the
component) but other than that it was working with accelerometer/compass -
LSM303, ALS's - NOA1305, microphones - MCP342x - most were using off the
shelf libraries from arduino but some i re-wrote as i only needed a tiny
subset of what they offered.

Paul

kriswiner, 2017-03-07:
> 284537786>,

@Paul This is with the nRF52?

Doesn't work for me.

XaviLC, 2017-03-08:
@nihaopaul How do you mean with _swap the two pins_ ?You're connecting SDA->SDA and SCL->SCL aren't you?

XaviLC, 2017-03-08:
I have connecte my adafruit sensor to pins 26 and 27, I see how the serial consolse of nrf52 gets stuck... Looking into the repository and  [I see  here ](https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/nRF52DK/variant.h) another pin definition of SDA/SCL, I tried them as well, the serial was not stuck but no data from my sensor though

XaviLC, 2017-03-08:
I have solved the issue, I've followed the comment of @nihaopaul  and I have connected my senor in this way
SENSOR (SDA)->P026(SCL nrf52)
SENSOR(SCL)->P0.27(SDA nrf52)

and starting the Wire service did the trick.

regards

dlabun, 2017-03-08:
That's great you figured out how to make it work. The big question is why do you need to roll the connections... Could either of you post a picture of your board? I don't have mine handy and I'd like to take a look at the silkscreen vs the datasheet.

XaviLC, 2017-03-09:
[image]
That is my set up, 
Blue is power ( I didn't have red :P)
Black is ground
Green SCL from the sensor
Yellow SDA from the sensor

nihaopaul, 2017-03-09:
> issues/116#issuecomment-
> 284698116>,

Yep both the nrf52dk and custom board, the nrf52 i have is V 0.9.0 pca10040
- i remember reading through the nordic release notes and it was
inconsistently listed under which pins to use, so i thought, what the heck,
and just wired them backwards and it worked :) i'm using the:

#include <Wire.h> library and also the #include <BLEPeripheral.h> library,
i also had to be specific on the clock src (my boards do not have a lfclk
src so i use the internal RC (when it said optional on the datasheet i
thought it was a good idea to leave it out, wish i didnt now) - i need to
revive the project though in the upcoming week and was based on development
branch not release at that time.

hope that helps

Paul

dlabun, 2017-03-09:
Hmmm... I'll take a look at the Nordic gerber files and see if they made an error in the routing of the PCB.

sandeepmistry, 2017-03-13:
@XaviLC 

> I have solved the issue, I've followed the comment of @nihaopaul and I have connected my senor in this way
> SENSOR (SDA)->P026(SCL nrf52)
> SENSOR(SCL)->P0.27(SDA nrf52)
>
> and starting the Wire service did the trick.

So SDA is pin 26 and SCL is pin 27 on the Nordic nRF52DK board. This lines up with the SDA and SCL pins on the Uno R3 board: https://i0.wp.com/makezine.com/wp-content/uploads/2013/02/arduino_uno_pinout_web.png?resize=1200%2C670

closed 2017-03-13 by sandeepmistry

XaviLC, 2017-03-14:
@sandeepmistry 

Yest that is correct, but as @dlabun commented it seems to be an issue with their datasheet since in their manual reference is stated that those pins are different, take a look into  pag 18

[nRF52_PDK_User_Guide_v1.2.pdf](https://github.com/sandeepmistry/arduino-nRF5/files/840636/nRF52_PDK_User_Guide_v1.2.pdf)
(url)

dlabun, 2017-03-14:
@sandeepmistry I think the issue is that the Preview version of the board contained either silkscreen or routing errors. It's hard for me to prove it at the moment since I have the production version of the Development Kit.

sandeepmistry, 2017-03-14:
@dlabun got it, I have the production version too.

Maybe we should use the Github wiki pages to store some board specific details (like I2S, SPI pins) so you don't have to dig through the variant files.

cc/ @jacobrosenthal