sandeepmistry/arduino-nRF5 #117: Does the use of LEDs need BLEPeripherial.h?

XaviLC, 2017-03-08:
Hello,
I was wondering whether or not switching on a LED will require the BLEPeripherial library.

I would like just use the EddySotne example of such a library but add that the LED2 fires up, I have used the followins snippet but the led is not turn on, so I was wondering whether it will related to the BLEPeripherial

> // LED pin
> #define LED_PIN   2 
> 
> void(){
>   // set LED pin to output mode
>   pinMode(LED_PIN, OUTPUT);}
>     digitalWrite(LED_PIN, HIGH);
> }

Thanks in advance,

dlabun, 2017-03-08:
No library is required and with the nRF52 development kit you don't need to define the LEDs as it's already been done in the core for you. The LEDS are named PIN_LED1, PIN_LED2, PIN_LED3 and PIN_LED4.

sandeepmistry, 2017-03-13:
@XaviLC as @dlabun mentioned above no library required.

Also the built-in LED's on the nRF52DK board are inverse from what i remember: `LOW` will actually turn the LED on.

closed 2017-03-13 by sandeepmistry

XaviLC, 2017-03-14:
@dlabun and @sandeepmistry  Thanks, I will try so, could you point me out where the LED definitions are?

jacobrosenthal, 2017-03-14:
Assuming in this case say nrf52dk, Theyre defined here:
https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/nRF52DK/variant.h#L48-L51