sandeepmistry/arduino-nRF5 #118: Linking Error using s132_nrf52_3.1.0_softdevice.hex

bossesand, 2017-03-10:
I cannot use merge hex with this  softdevice using mergehex
Which version is to be used?

using JFlash  S132 uses addresses 00000 - 1EBEF
But the compiled binary is compiled for 1C000 this is a conflict

Am I using a to new Softdevice ?

bossesand, 2017-03-10:
s132_nrf52_2.0.1_softdevice seems to end at 1B6C7  so I guess I was too eager to use a newer version of the softdevice.

Is it "ONLY" needed to replace the .h files or where is the destination base address for the compiled program specified?

Best Regards
Bo-Erik

sandeepmistry, 2017-03-13:
@bossesand 

Here are the places i recall:

* https://github.com/sandeepmistry/arduino-nRF5/blob/master/softdevices.txt#L9
* https://github.com/sandeepmistry/arduino-nRF5/tree/master/cores/nRF5/SDK/components/softdevice/s132/headers
* https://github.com/sandeepmistry/arduino-nRF5/blob/master/cores/nRF5/SDK/components/softdevice/s132/toolchain/armgcc/armgcc_s132_nrf52832_xxaa.ld

sandeepmistry, 2017-04-24:
Closing this for now, as I think the question has been answered.

closed 2017-04-24 by sandeepmistry