sandeepmistry/arduino-nRF5 #119: Is possible to configure pins in DK as UART?

XaviLC, 2017-03-14:
Hello,

I was wondering whether or not it would be possible to convert some of the available pins of nrF52 DK into another UART. I would like to connect a device that communicates through UART and I was considering to use two of the available pins. Should I map thos pins in the [variant.h](https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/nRF52DK/variant.h) for the nr52DK? Or is there a better way of doing so?

THanks in advance,

REgards!

dlabun, 2017-03-14:
Sorry, but the nRF series only have a single UART so you cannot add another. There are I2C/SPI to UART bridge chips on the market you may want to look at.

closed 2017-03-14 by dlabun

XaviLC, 2017-03-14:
OK @dlabun  thanks!

jacobrosenthal, 2017-03-14:
Not sure I fully agree.

You can add another uart on any pins on an arm chip like nrf51 or nrf52.
Though that said it wont be able to talk to your laptop on say your usb
without something like an ftdi chip or other microcontroller turning that
uart into a virtual usb serial driver.

dlabun, 2017-03-14:
Per the Nordic datasheets, there's only a single UART transceiver built into the nRF family. Chips like the Atmel SAMD21 used on the Arduino Zero do include multiple UARTs but Nordic doesn't.

jacobrosenthal, 2017-03-14:
Ah, interesting, thanks @dlabun You can put it anywhere you want, but yeah
only one set of associated timers/registers.

Though you CAN change the pins while running though to switch to another
device, assuming you assert RTS/CTS or dont care if you miss some data
during the switch.

OR youd have to find/build a bitbang/softserial library to turn random gpio
into uart.

On Tue, Mar 14, 2017 at 12:07 PM, dlabun <notifications@github.com> wrote:

> Per the Nordic datasheets, there's only a single UART transceiver built
> into the nRF family. Chips like the Atmel SAMD21 used on the Arduino Zero
> do include multiple UARTs but Nordic doesn't.
>
> —
>