sandeepmistry/arduino-nRF5 #120: How can I configure an interruption?

XaviLC, 2017-03-15:
Hello all,

I was wondering, can I make use of the interruptions for the nrf52 dk? 

 I would like to attach some sensor(accelerometer) into my board and I would like to configure that it can interrupt if there is a movement, basic stuff; the library that I am using for it doesn't have implemented the interruption itself so I will have to code it for the nrf52, is ther any example that I might use? 

Thank in advance,

regards!

carlosperate, 2017-03-25:
The `attachInterrupt` arduino function (https://www.arduino.cc/en/Reference/attachInterrupt) is already implemented in `cores/nRF5/WInterrupts.h/.c`.

sandeepmistry, 2017-03-27:
@carlosperate thanks for helping out!

@XaviLC please see the answer above.

closed 2017-03-27 by sandeepmistry