sandeepmistry/arduino-nRF5 #121: Does the library support timers or watchdog?

XaviLC, 2017-03-15:
Hello 

I've been looking through the repo looking for something similar to arduino RTCTimer or RTCZero libraries, is there a way that I can use similar libraries or functions? I would like to set up some timers that might make my nrf52 be woken up after certain amount of time.

Thanks in advance,

XaviLC, 2017-03-21:
Hi @sandeepmistry  or @dlabun  could you tell me if the current status of this lib support the nordic drivers such as RTC driver or the watchdog driver ?

Thanks

d00616, 2017-03-24:
@XaviLC If you need some watchdog code, you can use my implementation of avr/wdt.h from https://github.com/sandeepmistry/arduino-nRF5/pull/55.

dlabun, 2017-03-26:
There is no real time clock on nRF chips like the Arduino Zero has.  As far as I know support for the Nordic watchdog has not been implemented yet.

carlosperate, 2017-03-26:
> There is no real time clock on nRF chips like the Arduino Zero has.

I'm not familiar with the Arduino Zero RTC capabilities, but the nRF micros do have a Real Time counter and if RTC oscillator is not included it can "synthesise" a 32.768 kHz clock from the HFCLK. Is there something else required that is missing?

dlabun, 2017-03-26:
The Arduino Zero libraries you mention are for time keeping using the internal real time clock. There really isn't anything similar for the nRF.

carlosperate, 2017-03-27:
Just to clarify that I wasn't implying there are is RTC library for the nRF51/2 available at the moment, just that these microcontrollers do have a Real Time Counter that could be used for this purpose (I though you were implying otherwise in your previous message, but I might have misinterpreted it).
Even if this counter is already in use to provide the `millis()` and `micros()` functions, it could be configured to do both, no?

sandeepmistry, 2017-03-27:
@XaviLC 

> could you tell me if the current status of this lib support the nordic drivers such as RTC driver or the watchdog driver ?

Nothing is present currently, pull requests welcome.

@carlosperate 

>Even if this counter is already in use to provide the millis() and micros() functions, it could be configured to do both, no?

Right, RTC1 is fixed for 1ms ticks, RTC0 is reserved for the soft device

Regular timers could be based on Paul's TimerOne library: https://github.com/PaulStoffregen/TimerOne - the CurieTimerOne library for the 101 was based on this.

I'm going to close this for now, as the original question was answered.

closed 2017-03-27 by sandeepmistry