sandeepmistry/arduino-nRF5 #129: Tone library not implemented

carlosperate, 2017-03-27:
Looks like the Tone library has not yet been implemented, which is a standard Arduino library also included in the other official cores.
Has anybody looked into this before?

sandeepmistry, 2017-04-10:
@carlosperate I'm not planning to look into this.

It would awesome if you or someone else could submit a pull request for this functionality.

carlosperate, 2017-04-10:
At the moment I won't be able to dedicate the time to port this one, but it's certainly something I'd be interested to have, so happy to help somebody else if anybody would like to give it a go.
Depending on other stuff I might have to push this one higher or lower in my to-do list.

scientistnobee, 2018-08-29:
Does anyone ported this library? Please let me know. Thanks.