sandeepmistry/arduino-nRF5 #130: Clarification on PWM/analogWrite ?

carlosperate, 2017-03-28:
This might be more of a general question, the analogWrite function has the following comment:

```
// Right now, PWM output only works on the pins with
// hardware support.  These are defined in the appropriate
// pins_*.c file.  For the rest of the pins, we default
// to digital output.
void analogWrite( uint32_t ulPin, uint32_t ulValue )
```

I assume this is referring to the `NUM_ANALOG_OUTPUTS` macro, but all the variants have this macro set to `(0u)`, and it is not used anywhere in the code.

On the other hand the wiring_analog_nRF51.c and wiring_analog_nRF52.c files do have the following macro defined and in use:
```c
#define PWM_COUNT 3
```

So is there any correlation between these two? or I am getting confused with something unrelated?

Apart from that, the timers have 4 compare registers, so I was wondering if there is a reason this has been restricted to only 3?

sandeepmistry, 2017-04-10:
`NUM_ANALOG_OUTPUTS` is for DAC's, and part of the Arduino standard.

On nRF52, hardware PWM's are used so that's the limit. nRF51 can probably have more than three, I forget why I capped it at 3, but am open to change it.

Closing this for now, because i think the question is answered, but feel free to continue the discussion.

closed 2017-04-10 by sandeepmistry

carlosperate, 2017-04-10:
Thanks @sandeepmistry , this answers my question. We can always revisit expanding the current analogWrite limit if necessary.