sandeepmistry/arduino-nRF5 #133: analogRead() should be able to map inputs 0-to-x as A0-to-Ax

carlosperate, 2017-04-03:
The default Arduino analogRead() can take 2 different values for each ADC input:
1. The "analog pin" value, which goes from 0 to NUM_ANALOG_INPUTS, which in most Arduino Uno compatible boards would be 0 to 5:
    ````arduino
    analogRead(0);
    analogRead(1);
    ...
    analogRead(5);
    ````
2. The pin number, in it's "integer form" or `Ax` global variable:
    ````arduino
    analogRead(A0);
    analogRead(A1);
    ...
    analogRead(A5);
    ````
    Which in boards like the Uno would be the equivalent of:
    ````arduino
    analogRead(14);
    analogRead(15);
    ...
    analogRead(19);
    ````

This is usually done by simply checking the range of the input argument, so in the case of the Uno the `analogRead()` function would have something along the lines of:

```arduino
analogRead(uint8_t pin) {
  if (pin < A0) {
    pin += A0;
  }
}
```

That works well because these boards are configured to have the analog pins mapped immediately after the digital pins and in successive order.

This Arduino core has variants with analog pin mappings that do not follow this convention, and so only "method 2" listed above has been implemented.

This breaks compatibility with any arduino sketches using "method 1", so we should work out a method to map these analog pins in the variant files in a way that allows both methods to work as expected.

sandeepmistry, 2017-04-10:
@carlosperate I agree, could you submit a PR for this?

Maybe we can have a `#define` for non-compatible variants - the ones with non-Arduino like pin definitions to disable this option. What do you think?

carlosperate, 2017-04-10:
Maybe I can put something together in a couple of days, depends on availability, and how much work left there might be for the other PRs.

I think we need to do a pin conversion similar to `if (pin < A0) pin += A0;`, but since this would be specific to each board, a macro/function would have to be defined per variant.

So, in essence, if the input is `< NUM_ANALOG_INPUTS`, then we'd apply this "variant dependant" macro/function, and if not then just continue as it is.

sandeepmistry, 2017-04-10:
I was thinking something super simple like:

```c
#ifndef ARDUINO_NON_STANDARD_ANALOG_PINS
if (pin < A0) {
  pin += A0;
}
#endif
```

Then the non-standard variants like the generic, TinyBLE, nRF51 beacon (in the future) can have a `#define ARDUINO_NON_STANDARD_ANALOG_PINS` in the variant header. What do you think? I can prep a PR if this is ok.

carlosperate, 2017-04-10:
Yes, something like that would definitely work, but then it would leave the variants with a "non-standard analog pin distribution" with the same problem as before.

We could do this, in the variant file:

```cpp
#define ARDUINO_NON_STANDARD_ANALOG_PINS
static __inline__ uint32_t AnalogPinConver(uint32_t pin) {
    // map pins
    return pin;
}
```
And analogWrite:
```cpp
#ifdef ARDUINO_NON_STANDARD_ANALOG_PINS
pin = AnalogPinConver(pin);
#else
if (pin < A0) {
  pin += A0;
}
#endif
```

sandeepmistry, 2017-04-10:
That's a good suggestion!

To confirm,  `AnalogPinConvert(0)` returns the pin associated with A0?

@dlabun @jacobrosenthal any opinions on this?

carlosperate, 2017-04-10:
yeah, the `AnalogPinConver()` function would have a lookup table for that conversion, so as you say input '0' would return the pin set for 'A0'.

carlosperate, 2017-04-21:
The proposed solution discussed above can be found at #150.

micooke, 2017-09-12:
I dont get it, is this actually needed? How is this not covered by the variable mapping in variants.h