sandeepmistry/arduino-nRF5 #134: Nordic PAN 56: TWI module lock -up

carlosperate, 2017-04-05:
I've been trying to debug some issues I am having with the I2C driver (I'll submit an unrelated PR for that soon), and googling around I've encountered quite a few mentions of this Product Anomaly Notice # 56 for nRF51822.

The following conditions will make the TWI lock up: 
* Disabling the TWI module while the TWI slave clock-stretches an ongoing read or write sequence.
* Triggering the RESUME task too early after receiving RXDRDY event. Firmware triggers the RESUME task 
within the same CPU clock cycle as the ACK bit is finished sending from the TWI master

This only affects the silicon v2.4 or earlier (circa 2014), and while I assume v3.x is fairly common these days, going through my boards I've found at least one that would be affected. Since the software reset workaround is not that bad, it's probably not unreasonable to include it.

http://infocenter.nordicsemi.com/pdf/nRF51822-pan_v2.4.pdf

dlabun, 2017-04-05:
It sounds like it'd take some work to really get to the point of locking up the TWI module... Personal opinion, I don't think this really needs to be built in but it's good to know about for the fringe cases.

jacobrosenthal, 2017-04-10:
@carlosperate  do you know which variants we already define might be susceptible to this?

carlosperate, 2017-04-10:
Well, I assume most of the nRF51 variants use the nRF51822 or nRF51422, so technically most if not all of the nRF51?  It's a matter when "when" were the boards built, as newer batches might contained the newer silicon version with the bugfix.

sandeepmistry, 2017-04-23:
Closed via https://github.com/sandeepmistry/arduino-nRF5/issues/137.

closed 2017-04-23 by sandeepmistry