sandeepmistry/arduino-nRF5 #135: BLEPeriferal serial example - cannot connect using Nordic -UART app

bossesand, 2017-04-05:
I am trying to make device that reads a IMU over twi/i2c and sends the data over ble uart

I take it step by step to establish the BLE UART, iI am using the serial example in BLEPeriferal lib S132 is also loaded
I have the Taida Century nrf52 mini board

I cannot get the example code to accespt a connect using the nrF UART 2.0 App.
It just hang in the connect stage.

I have modified the code to send the spam timestamps
I ahve changed the BLE_RDY to another pin, I have not found any simple explanation on how to use these BLE_REQ BLE_RDY BLE_RST pins, do they have any 
interaction with hardware or are they "just" using resister?

Cany any one explain whats going on?

#define BLE_RDY   8

// Import libraries (BLEPeripheral depends on SPI)
#include <SPI.h>
#include <BLEPeripheral.h>
#include "BLESerial.h"

// define pins (varies per shield/board)
#define BLE_REQ   10
#define BLE_RDY   8
#define BLE_RST   9

// create ble serial instance, see pinouts above
BLESerial BLESerial(BLE_REQ, BLE_RDY, BLE_RST);

void setup() {
  // custom services and characteristics can be added as well
  BLESerial.setLocalName("OHTI-test");

  Serial.begin(115200);
  BLESerial.begin();
}

void loop() {
  BLESerial.poll();

  // forward();
  // loopback();
  spam();
}

// forward received from Serial to BLESerial and vice versa
void forward() {
  if (BLESerial && Serial) {
    int byte;
    while ((byte = BLESerial.read()) > 0) Serial.write((char)byte);
    while ((byte = Serial.read()) > 0) BLESerial.write((char)byte);
  }
}

// echo all received data back
void loopback() {
  if (BLESerial) {
    int byte;
    while ((byte = BLESerial.read()) > 0) BLESerial.write(byte);
  }
}

// periodically sent time stamps
void spam() {
  if (BLESerial) {
    BLESerial.print(millis());
    BLESerial.println(" tick-tacks!");
    delay(1000);
  }
}

dlabun, 2017-04-05:
The Nordic UART application is intended to be used with the Nordic specific example code. It's very difficult to tell what your issue is since the Nordic UART is not support here.

sandeepmistry, 2017-04-10:
Closing this and a duplicate issue was created here: https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/136

closed 2017-04-10 by sandeepmistry