sandeepmistry/arduino-nRF5 #136: Low Power modes

kriswiner, 2017-04-05:
Is there a plan to enable low power settings on the nRF52? I would like to be able to put the BLE device in the lowest power settings when not actively sending out data. Any idea when this might be available?

dlabun, 2017-04-06:
Power management isn't available in any of the official Arduino cores, it's pretty much implemented on an as needed basis by advanced users. It's probably not a high priority item at this time.

sandeepmistry, 2017-04-10:
There's a new Arduino Low power library we can port over as a bundled library: https://github.com/arduino-libraries/ArduinoLowPower

kriswiner, 2017-04-10:
Yes, I saw that, not sure this will save much power with the nRF52 without
a specific nRF52-based API.

kriswiner, 2017-04-10:
I'd be happy to gove it a try though. There is no RTC on the nRF52, just a
real time counter. But having either timed sleep or wake on interrupt sleep
might allow a successful low-power strategy to be devised.

I was able to get the average power down from ~1 mA to 150 uA with the
STM32L4 running some I2C sensors and displaying their data on a TFT every 5
seconds using something similar so it is worth a try with the nRF52, which
is something of a power hog without any power management.

jacobrosenthal, 2017-04-10:
@kriswiner are you talking about dcdc mode? or crystal selection?

kriswiner, 2017-04-10:
Neither. Low power modes for the nRF52 cannot be compatible with low power
modes of the SAMD unless by accident. I mean how to suspend the CPU, which
peripherals remain active, which clocks can be used in which modes, and
which events can wake and into which state the device can wake into, etc
have to be specifically tailored for the nRF52. Even for the generic
low-power mode "sleep" I expect there will have to be some changes to the
API for the nRF52, that's all. It is also unlikely this simple low power
"library" will allow full power management of the nRF52 in real
applications. But I am willing to suspend my disbelief long enough to give
it a try and measure how well it does.

I had to use STM32L4 STOP modes in combination with setting the "runl" CPU
clock speed to 1 MHz in order to get the power usage down to 150 uA. I
expect something similar will be required for the nRF52, but I do not think
there is any provision in the Arduino code (or maybe the Nordic SDK) for
changing nRF52 CPU clock speeds and I don't know to which nRF52 mode
"sleep" corresponds.

jacobrosenthal, 2017-04-10:
Thats exactly what the low power library sandeep linked is for then, though you're right it needs to be altered and pull requested here for nrf51 and nrf52 registers.

dlabun, 2017-04-10:
Most sporting / fitness equipment that uses the ANT protocol use these Nordic chips... I'd have to imagine there's a standardize sleep or low power mode. I know the sensors on my bicycle are nRF based and the batteries last a long time.

kriswiner, 2017-04-10:
"standardize sleep or low power mode"

I doubt it, the way this low power usage is achieved is by tri-stating all
of the GPIOs not actively being used, shutting down all the peripherals not
actively being used, shutting down the radio when it is not actively being
used, and shutting down the CPU when it is not actively being used. Then
efficient activation of the above when they are required by the program and
transition from low power to usable states in such a way to avoid locking
up the whole works. This requires detailed knowledge of the workings of the
underlying MCU; it is unlikely simply using a lowpower.sleep command
pinched from a SAMD API will accomplish all of this specific management.

I would be delighted to be proven wrong though.

jacobrosenthal, 2017-04-10:
@kriswiner Correct, theres nothing register wise from the samd you can copy for use here -- Were pointing out that api is the arduino way to get what you want. Now somebody (you?) has to do all the work you mentioned.

kriswiner, 2017-04-10:
No, not me.

dlabun, 2017-04-10:
@kriswiner Go look at the Nordic SDK, power management and scheduling is built in. Seems like plenty of professional companies have solved the low power issue.

kriswiner, 2017-04-10:
Not within an Arduino core though;>

jacobrosenthal, 2017-04-10:
Well all just have to keep waiting then..

closed 2017-04-10 by jacobrosenthal

curtpw, 2017-06-18:
Apparently there is now a low power library for the Primo (nrf52) from the folks at arduino.org. Not sure how difficult it would be to get that working with this repo: https://github.com/arduino-libraries/ArduinoLowPower/blob/master/examples/PrimoDeepSleep/PrimoDeepSleep.ino

TamojitSaha, 2019-12-04:
I know this a very old thread. But in case, if anyone is still struggling to find a library for NRF5x low power modes, [here](https://github.com/mristau/Arduino_nRF5x_lowPower) is one.