sandeepmistry/arduino-nRF5 #141: SoftwareSerial.h not found

OwenBrotherwood, 2017-04-08:
Trying to use a sketch with SoftwareSerial.h: but get a:
`C:\Users\owen\Desktop\rn\rn.ino:17:28: fatal error: SoftwareSerial.h: No such file or directory
`
The same sketch can be compiled if the board is an Uno.

Anything I can try to get further with this problem?

```
Detecting libraries used...
"C:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10802 -DARDUINO_NRF52_DK -DARDUINO_ARCH_NRF5  -DNRF52 -DNRF5 "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/CMSIS/Include" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/gcc/" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/device/" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/drivers_nrf/delay/" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/softdevice/none/headers/" -DUSE_LFXO "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\variants\nRF52DK" "C:\Users\owen\AppData\Local\Temp\arduino_build_220437\sketch\rn.ino.cpp" -o "nul"
"C:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10802 -DARDUINO_NRF52_DK -DARDUINO_ARCH_NRF5  -DNRF52 -DNRF5 "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/CMSIS/Include" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/toolchain/gcc/" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/device/" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/drivers_nrf/delay/" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5/SDK/components/softdevice/none/headers/" -DUSE_LFXO "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\cores\nRF5" "-IC:\Users\owen\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.3.0\variants\nRF52DK" "C:\Users\owen\AppData\Local\Temp\arduino_build_220437\sketch\rn.ino.cpp" -o "C:\Users\owen\AppData\Local\Temp\arduino_build_220437\preproc\ctags_target_for_gcc_minus_e.cpp"
C:\Users\owen\Desktop\rn\rn.ino:17:28: fatal error: SoftwareSerial.h: No such file or directory

compilation terminated.
```

https://github.com/sandeepmistry/arduino-nRF5/pull/54#issuecomment-258714787

dlabun, 2017-04-08:
Sorry, but soft serial ports are not currently support on the nRF chips.

closed 2017-04-08 by dlabun

sandeepmistry, 2017-04-10:
https://github.com/sandeepmistry/arduino-nRF5/issues/98 is open to track the request for a SoftwareSerial lib.