sandeepmistry/arduino-nRF5 #146: New firmware not running?

fpereiracostaa, 2017-04-11:
Hi, im trying to run the iBeacon example, but i cant detect it. 

It says Done uploading and I get this message:

`Open On-Chip Debugger 0.10.0-dev-gdc53227 (2016-04-09-13:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
0x4000
adapter speed: 1000 kHz
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x000001f0 msp: 0x20008000
** Programming Started **
auto erase enabled
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000001e msp: 0x20008000
wrote 15360 bytes from file /var/folders/lp/s4tcvj2x0ss5glz6y7q9w3mm0000gn/T/arduino_build_681942/iBeacon.ino.hex in 0.713888s (21.012 KiB/s)
** Programming Finished **
** Verify Started **
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20008000
verified 14464 bytes in 0.106868s (132.172 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked`

But still cant find the beacon.

Do i need to add more wires after upload? like ENABLE or smothing? Im using 4 wires, Vcc, GND, SWCLK and SWDIO from my ST-LINK v2.

Thanks a lot.

dlabun, 2017-04-12:
Where did you get the iBeacon example from? The log above indicates it flashed successfully.

fpereiracostaa, 2017-04-12:
From this recommended library:
https://github.com/sandeepmistry/arduino-BLEPeripheral
The iBeacon example.

carlosperate, 2017-04-12:
What board are you using? Could you get a simple LED blinking first?

sandeepmistry, 2017-04-13:
@fpereiracostaa have you flashed a soft device first? https://github.com/sandeepmistry/arduino-nRF5#flashing-a-softdevice

sandeepmistry, 2017-08-11:
Closing for now due to lack of activity.

closed 2017-08-11 by sandeepmistry

kartik9k, 2018-05-30:
Hi! I'm facing a similar issue while uploading. I've uploaded an example sketch on my Redbear BLE Nano 2. The upload logs are exactly the same but it does not show up as a peripheral or a beacon.

Is there any fix for this? 
TIA