sandeepmistry/arduino-nRF5 #148: Is this works with the Sparkfun breakout?

Curnelius, 2017-04-18:
this one : https://www.sparkfun.com/products/13990  
Can you use the Nordic flash to save data ?
Thank you! seems like a great library .

carlosperate, 2017-04-21:
There is no specific target for this board in particular, but I don't see why it woudn't work with the default nRF52832 variant.

However, keep in mind that this board has been designed with the idea of using the internal bootloader (which is installed from factory) and flash a program through this bootloader (looks like you also need a USB-to-serial converter for that). The Arduino core in this repository is designed to flash the boards directly using a programmer and without including a bootloader.

Curnelius, 2017-04-22:
Yes I know that thanks very much for pointing that out :)
This seems great, I wonder if there is a way to get access to the nrf flash to save data using an Arduino library ? 
Also, technically , working with Arduino- will I have a lower performance ?

carlosperate, 2017-04-22:
I don't know if there is any Arduino flash library for the nRF5 architecture, but it is definitely possible to write to flash from a running program, however keep in mind that the block size is 1Kb and each block needs to be deleted before any data can be written into it.

> Also, technically , working with Arduino- will I have a lower performance ?

Lower performance compared to what?

Curnelius, 2017-04-23:
Data sheet says 512K of Flash , why you say the block is only 1K ? or did you mean many blocks?

Lower compared to writing direct code using their native SDK and access directly to registers, etc .

carlosperate, 2017-04-23:
Sorry I didn't use that term correctly, for the nRF51 (the architecture I was mistakenly thinking of when I wrote my previous message) flash only has pages, no blocks, and each page is 1Kb.
For the nRF52 in this case flash is divided in pages of 4Kbs and each page is then divided in blocks of 512B.
For both micros, you have to erase before you write, and the smallest unit you can erase is a page.

sandeepmistry, 2017-04-23:
I'm closing this as answered, see @carlosperate's answer in https://github.com/sandeepmistry/arduino-nRF5/issues/148#issuecomment-296329458 - but feel free to continue the discussion.

> Lower compared to writing direct code using their native SDK and access directly to registers, etc .

@Curnelius I would expect equivalent results to the native SDK, as this core accesses the registers directly just like the native SDK. If you spot any performances please open an issue or PR to discuss the specifics.

closed 2017-04-23 by sandeepmistry