sandeepmistry/arduino-nRF5 #149: Serial not working on generic nRF52

vanbwodonk, 2017-04-19:
Hi, thanks a lot for making this core. It's great.
I'm buy cheap nRF52832  from china and i suggest it's nRF52 generic on this core. Here's link i buy and pinout on the web too. https://goo.gl/ZsTRXg.
I'm successfully make blinky code so i believe this core is working. But after trying serial code, `Serial.print()` can't send any data. I'm checked data serial output from PIN 1(from variant.h).

vanbwodonk, 2017-04-19:
After some trial, my serial working after i change pin (in variant.h) from
```
#define PIN_SERIAL_RX       (0)
#define PIN_SERIAL_TX       (1)
```
into
```
#define PIN_SERIAL_RX       (6)
#define PIN_SERIAL_TX       (7)
```
I'm not sure but i think 32KHz crystal connect to P0 and P1. Maybe we need new variant for this board.

carlosperate, 2017-04-21:
The UART can be mapped to pretty much any unused pin, is there anything in the board connected to pin 0 or 1? If so then yes, it would probably need its own variant (although maybe other nRF52 variants already present might work as well).

Do you know if there is an schematic available?

vanbwodonk, 2017-04-22:
@carlosperate I don't think they will release schematic. But now i'm pretty sure P0 and P1 connect into 32KHz chystal.

carlosperate, 2017-04-22:
Yes, I forgot those are inputs can also be used for the about the low frequency crystal.

@sandeepmistry perhaps this is something that should be addressed in the generic variant?
If the USE_LFXO flag is set then the serial pins should be set to a something else.
And maybe for all variants if USE_LFXO is set, we should probably remove the respective pins (0 and 1 in nRF52 and 26-27 in nRF51) from use by anything else and map them out of `g_ADigitalPinMap[]`.

sandeepmistry, 2017-04-23:
@carlosperate your suggestion sounds good to me.

@vanbwodonk thanks for the positive feedback! I'd recommend you to make a custom variant for your board. While we wait for a PR for the suggestion above (or maybe you can submit a PR for it?)

Jepeto4, 2017-10-24:
Hello,
I am adding the schematic in case you still need it. Perhaps I am missing something here, but I wonder why why we would not simply define the TX and RX pins as P06 and P08 as they are laid out in nRF52?
[image]

sandeepmistry, 2018-01-14:
There's a new `setPins` API in PR https://github.com/sandeepmistry/arduino-nRF5/pull/221 but @honnet to select custom pins for the generic variant now.

closed 2018-08-19 by sandeepmistry

honnet, 2018-08-19:
Selamat @vanbwodonk ;)

Here is how to use the function:
```
Serial.setPins(RX, TX); // must be before Serial begin() call!
Serial.begin(BaudRate);
```

More details here: https://github.com/sandeepmistry/arduino-nRF5/pull/221/commits/9ae9453c68606398be44c05cc92c022f7a388969

Cedric ;)

pvonmoradi, 2021-01-04:
@honnet  Is this PR merged?