sandeepmistry/arduino-nRF5 #153: attachInterrupt() current consumption

d00616, 2017-04-26:
When the MCU goes into sleep mode after attachInterrupt() it consumes 1mA. This is too much for a battery powered application.

I prefer to switch from Config Channels to Pin Sense. This allows sleeping with uA current. Pin Sense is not so flexible like Config Channels. This requires a logic to find out what pin was changed. Falling and Rising detection is easily possible. Pin change needs some logic. 

Switching to Pin Sense allows monitoring of more Pins than Config Channels.

If it not possible to switch to pin sense, I need handling of NRF_GPIOTE->EVENTS_PORT, so I can implement my own routine.

When it is acceptable to switch to Pin Sense, I create a Pull Request.

Btw. I think the actual implementation will fail when multiple pins are inserted and removed in different order.

sandeepmistry, 2017-05-08:
@d00616 how are you making the board go to sleep?

d00616, 2017-05-08:
@sandeepmistry I use the `__WFE(); __SEV(); __WFE();` sequence.

When GPIOTE channels are disabled, I measure a sleeping current around 3uA. When enabled the current is around 950uA. -> https://devzone.nordicsemi.com/question/2471/gpio-wakeup-low-power-workarounds/

d00616, 2017-07-02:
Hello @sandeepmistry it seems the high current consumption is a nRF51 specific problem documented as PAN #39. It's marked as fixed in PAN v 3.0 and later: http://infocenter.nordicsemi.com/pdf/nRF51822-pan_v3.0.pdf

closed 2017-07-02 by d00616

jeanmatthieud, 2019-06-27:
Hi,
It's not a problem, it's a known limitation of the chip, cf. PS 8.11 :

> Setting up one or more GPIO DETECT signals to generate PORT EVENT, which can be used either as a wakeup source or to give an interrupt, will not lead to an increase of the current consumption.

I agree with the initial issue: with the current implementation, it's not possible to switch to pin sense (sometimes called "low accuracy") and get the callback from the interrupt.