sandeepmistry/arduino-nRF5 #154: genneric nrf52832 i2c

reapqin, 2017-04-27:
@kriswiner i am working with nrf52832 + mpu9250 also. And i see your [commet](https://github.com/sandeepmistry/arduino-nRF5/issues/108) about the "I switched to the nRF51Wire.h and everything works"
could you tell me how to switch that? I am begginer wish your help.

i am using a genneric [nrf52832](https://item.taobao.com/item.htm?spm=a230r.1.14.74.IuClmR&id=532630623264&ns=1&abbucket=15#detail)  with all pins out around borad and named by original pin name on chip datasheet. and i changed the sda to 26u scl to 27u in variant.h in genneric fold
and i found the board pins 26 27 was not pullup with resistors. i don't have 4k7 resistors so i use 5k1 resistors instead. is it neccesory?
Wish your help thank you.

kriswiner, 2017-04-27:
Yes, you need pullups either on your nRF52 board or MPU9250 breakout.

reapqin, 2017-04-27:
And then ?  how  to  switch  to  nrf51wire.h? thank you.

carlosperate, 2017-04-27:
The 13 KOhm internal pull ups could be enabled for the I2C pins (they are within this Wire library), so technically speaking external pull ups could not be required.

kriswiner, 2017-04-27:
You'll have to test this; i use 4K7 pullups.

reapqin, 2017-04-28:
@kriswiner  ok,  i find 4k7 resistor  and pulled  up. what's next?  thanks ₍ↂ⃙⃙⃚⃛_ↂ⃙⃙⃚⃛₎

kriswiner, 2017-04-28:
run the code

reapqin, 2017-04-28:
@kriswiner  it's  not  work.

kriswiner, 2017-04-28:
open an issue

reapqin, 2017-04-29:
I test that. it is not neccesory to pullup.

dlabun, 2017-05-14:
Closing as resolved (?)

closed 2017-05-14 by dlabun