sandeepmistry/arduino-nRF5 #155: NFC pins as GPIO

clduce, 2017-04-27:
I'm working with a custom nRF52832  board. I'm using the "Generic nRF52" board setting.
I'm trying to use pins 9 and 10 as general purpose output pins, but they are not working. I know these pins are the NFC pins, and have read that they need to be set as GPIO in the UICR before they will work, but I don't know how to go about doing that.
All the other GPIO pins work as expected when I use a simple program to toggle them. 
I know this isn't a hardware issue such as unsoldered pin etc, since I have the same issue on 10 different boards.
How do I go about enabling pin 9 and 10 for GPIO?

Thanks,
Cory

sandeepmistry, 2017-05-08:
@clduce does the nRF52832 data sheet mention they can be used for GPIO?

clduce, 2017-05-08:
Yes they can be. I've read others doing it, but they are using the Keil IDE.
For example, this thread
[https://devzone.nordicsemi.com/question/55696/struggling-with-basic-gpio-read-on-nrf52/]
suggests making changes to the UICR register, but I'm not sure how to do that using Arduino and the nRF5 core.

sandeepmistry, 2017-05-15:
@clduce I suggest you post on the dev zone: https://devzone.nordicsemi.com/questions/

and ask how you can set the UICR registers to NFC GPIO mode in C.

mlu, 2017-07-08:
There is code in the SystemInit() function in the file system_nrf52.c   line 141
I took some code from there and placed in my setup function:

`/* Configure NFCT pins P9 and P10 as GPIOs if NFCT is not to be used in your code. If CONFIG_NFCT_PINS_AS_GPIOS is not defined,
   two GPIOs (see Product Specification to see which ones) will be reserved for NFC and will not be available as
   normal GPIOs. */

  if ((NRF_UICR->NFCPINS & UICR_NFCPINS_PROTECT_Msk) == (UICR_NFCPINS_PROTECT_NFC << UICR_NFCPINS_PROTECT_Pos)){
      NRF_NVMC->CONFIG = NVMC_CONFIG_WEN_Wen << NVMC_CONFIG_WEN_Pos;
      while (NRF_NVMC->READY == NVMC_READY_READY_Busy){}
      NRF_UICR->NFCPINS &= ~UICR_NFCPINS_PROTECT_Msk;
      while (NRF_NVMC->READY == NVMC_READY_READY_Busy){}
      NRF_NVMC->CONFIG = NVMC_CONFIG_WEN_Ren << NVMC_CONFIG_WEN_Pos;
      while (NRF_NVMC->READY == NVMC_READY_READY_Busy){}
      NVIC_SystemReset();
  }
`

mlu, 2017-07-08:
A simpler way is to add the flag in boards.txt (had to touch system_nrf52.c to rebuild the cached core ):

`Generic_nRF52832.build.extra_flags=-DNRF52 -DCONFIG_NFCT_PINS_AS_GPIOS
`

sandeepmistry, 2017-07-31:
@mlu nice find!

I'm closing this as resolved, maybe you can submit a pull request for this?

closed 2017-07-31 by sandeepmistry

drmpf, 2019-03-23:
Adding -DCONFIG_NFCT_PINS_AS_GPIOS  did now work for me until I modified the linker script to
armgcc_s132_nrf52832_xxaa.ld  to

```
/* Linker script to configure memory regions. */

SEARCH_DIR(.)
GROUP(-lgcc -lc -lnosys)

MEMORY
{
  FLASH (rx) : ORIGIN = 0x1c000, LENGTH = 0x64000
  RAM (rwx) :  ORIGIN = 0x20002080, LENGTH = 0xdf80
  UICR_NFCPINS(r) : ORIGIN = 0x1000120C, LENGTH = 0x04
}

SECTIONS
{
  .fs_data :
  {
    PROVIDE(__start_fs_data = .);
    KEEP(*(.fs_data))
    PROVIDE(__stop_fs_data = .);
  } > RAM
  
.uicrNfcPinsAddress : 
  { 
     KEEP(*(.uicrNfcPinsAddress)) 
  } > UICR_NFCPINS
  
} INSERT AFTER .data;

INCLUDE "nrf5x_common.ld"

```