sandeepmistry/arduino-nRF5 #156: Flash, production, and Nordic sdk

Curnelius, 2017-04-30:
Hi.
The library is great, and we are trying to understand how much can you depend on it for production?

1. Say I developed an application with Arduino IDE, what would be the procedure with mass production? can I produce a Hex/Make file that can be flashed with any Segger device ?

2. What about Flash memory API to save non volatile data ? (/like eeprom) , if there isn't one, how can you run an Nordic code (SDK) together with this library ? will it compile if I just use Nordic code ?

Thanks .

jacobrosenthal, 2017-04-30:
Yes you could upload the generated hex with anything, but many of the nrf devices are already defined using segger programmers (via openocd)

And Arduino has a headless command line mode for generation and programming so you could use that 
http://playground.arduino.cc/Learning/CommandLine

Curnelius, 2017-04-30:
Thanks, not sure I get you. Regarding flashing the device, you flash with J link Segger programmer anyway even with this library so I am not sure how its important.

Regarding the link, this is exactly what I am NOT looking for. I want to work inside the same IDE and execute all the endless libraries that Nordic has, and are not part of this library ( and probably will not be so you can't really work without them).

jacobrosenthal, 2017-04-30:
Maybe I dont understand you.

If you want to use the same non arduino ide(which?)  and all nordic
libraries, Im not sure why you're here using arduino-nRF5 at all then?

Curnelius, 2017-04-30:
I am here because its easy to use Arduino IDE instead of Eclipse.
Yet, I have a problem since you can't do with it almost nothing : (RTC,sleep,Flash storage, etc).
Hence, I was looking to use their sdk together Arduino IDE to get these things ..

prjh, 2017-05-02:
You can program you nrf52 device with a Nordic nrf52-DK build-in Segger via the JLink port for a external device via 10p JLink cable. 
For persistence, you can have a look to the [BLEPeripheral](https://github.com/sandeepmistry/arduino-BLEPeripheral) library at @sandeepmistry repo. At it's Bondstore implementation, you can see the usage of  the softdevices NVM functions. An alternative could be to combine a FRAM or EEProm chip memory.
OTA is also possible. Take the Nordic bootloader example with buildin DFU service from SDK-11 and adapt it to your HW schema (bootloader button pin). Using the Nordic tools you are able to merge your bootloader hex-file with the s132 softdevice. And you can build zip files for to create OTA packages for DFU using the Nordic nRf-Tool App.

sandeepmistry, 2017-05-08:
> Say I developed an application with Arduino IDE, what would be the procedure with mass production? can I produce a Hex/Make file that can be flashed with any Segger device ?

If you run "Sketch -> Export Compiled Binary" it will generate the .hex or .bin file without the soft device

> What about Flash memory API to save non volatile data ? RTC, OTA programing, all of these where not implemented and I wonder if you can use Nordic SDK code inside your sketch, if so, how ?

You'll have to access those features using the registers directly. Or you can make a new Arduino library with the Nordic SDK. @prjh also has a nice answer above.

closed 2017-05-08 by sandeepmistry