sandeepmistry/arduino-nRF5 #159: disable Softdevice S130 application interrupts(nrf51822)

cheeta1, 2017-05-10:
hi , i want to disable softDevice application interrupts for samll amount of time to execute a time senstive code and then re enable it. what will be the proper course of action you suggest? i tried the functions interrupts() and nointerrupts() but the bluetooth functionality is lost from these functions and the nrf51822 chip get stuck. suggest me how can i achieve my goal thanks

dlabun, 2017-05-10:
The soft device is written by Nordic, you should check with them on manipulating the interrupts.

sandeepmistry, 2017-05-15:
@cheeta1 yes, please ask on the Nordic Dev zone: https://devzone.nordicsemi.com/questions/

By default this core and the recommended BLE libraries don't enable soft device interrupts. So not to sure what you are doing.

closed 2017-05-15 by sandeepmistry

rixnco, 2017-08-18:
@cheeta1 I'm having the same problem. Did you find any answer? 

I'm using 
- a BLE nano and a BLE nano 2 
- sandeepmistry nrf5 boards support  0.3.0
- sandeepmistry BLEPeripheral 0.4.0
- An Adafruit 8x8 matrix ledbackpack thru I2C.

Without activating the BLE, I can communicate with the display without problems
When activating the BLE, display got corrupted.
Same results on both boards: nrf51 and nrf52

Thanks for any help / feedback.