sandeepmistry/arduino-nRF5 #161: export compiled nRF52832 hex from Arduino IDE then use in nRFGo Studio to hack read protected wearables

curtpw, 2017-05-14:
*I'm re-purposing a Chinese fitbit clone that uses the nRF52832 - but hardwire read protected
*I can program it just fine using mbed hex file, Segger J-Flash, nRFGoStudio and a Segger J-Link programmer (in SWD mode)
*In my experience, nRFGo Studio is the only thing that can program read protected nRF5X devices. I have hacked and programmed many read protected nRF51822 devices this way and have never been able to program them any other way.
* I don't like how mbed supports the nRF52. I want to use Sandeep's approach - it is well received. 
* Having installed the Sandeep nRF52 board etc. etc. and exported a simple blinky program using the S132 softdevice (as provided by Sandeep) but when I attempt to upload in nRFGo Studio I receive an error: "invalid SoftDevice start address is not 0x0"
**How can I fix this SoftDevice issue in nRFGo Studio?**
*** How can I use the Sandeepmistry nRF5X tools to program an nRF52 in nRFGo Studio?**
I  am posting this is bounty source with a reward. Please include software and file versions, links and other documentation if you can.
Thanks!!!

dlabun, 2017-05-14:
What you are asking for is dangerously close to, if not actually in violation of the Nordic Semiconductor licensing agreement for the soft device. I have edited and locked this post to prevent any legal blow back from Nordic... This is the type of thing that could get this project shutdown.

closed 2017-05-14 by dlabun

dlabun, 2017-05-14:
I have reached out to the OP via email to discuss further.