sandeepmistry/arduino-nRF5 #162: "invalid Softdevice start address" using Arduino/sandeepmistry hex file with nRFGo Studio (bounty) on nRF52832

turinturam, 2017-05-14:
*I'm using a Sparkfun nRF52832 breakout board (https://www.sparkfun.com/products/13990). 
*I can program it just fine using mbed hex file, Segger J-Flash, nRFGoStudio and a Segger J-Link programmer (in SWD mode).
*Because of my team it is essential that I use nRFGo Studio to program the breakout.
*I don't like how mbed supports the nRF52. I want to use Sandeep's approach - it is well received.
* I have installed the Sandeep nRF52 board in Arduino etc. etc. and exported a simple blinky program using the S132 softdevice but when I attempt to upload in nRFGo Studio I receive an error: "invalid SoftDevice start address is not 0x0" How can I fix this SoftDevice issue in nRFGo Studio? 

I am posting this is bountysource.com with a $15 reward. Please include software and file versions, links and other documentation if you would be so kind. Thanks!!!
~Turin

sandeepmistry, 2017-05-15:
If you merge the Softdevice hex and the exported hex file for the sketch together, you should be good to go.

closed 2017-05-15 by sandeepmistry

curtpw, 2017-05-16:
How do I make the index values of the exported application hex go higher so it doesn't conflict with the softdevice hex? Can't merge the two.

sandeepmistry, 2017-05-16:
For others following along, this seems resolved by: https://github.com/sandeepmistry/arduino-nRF5/issues/163#issuecomment-301686311