sandeepmistry/arduino-nRF5 #163: can't merge application hex with softdevice hex (using mergehex & s132_nrf52_2.0.1_softdevice.hex)

curtpw, 2017-05-15:
I'm using nRFGo Studio to program a Sparkfun nRF52832 breakout board. For reasons outside of my control I must use nRFGo Studio.

In order to program the nRF52 using application hex exported from Arduino IDE I must either merge the exported application hex with softdevice hex and load merged hex into nRFGo Studio OR program softdevice first and then program application hex. Neither work with Arduino Core exported hex BUT it does work when I select the nRF52 Primo board. But Sandeep's Arduino Core is way better than Primo!

When I try to merge Arduino Core application hex with softdevice in Nordic mergehex this is what I get (tried both directions):
`PS C:\Users\Nemo\Desktop\hex_work> mergehex  --merge arduino_nrf52_test.hex s132_nrf52_2.0.1_softdevice.hex --output test_combined.hex
Parsing input hex files.
Merging files.
ERROR: The hex files cannot be merged since there are conflicts.
PS C:\Users\Nemo\Desktop\hex_work> mergehex --merge s132_nrf52_2.0.1_softdevice.hex arduino_nrf51_test.hex --output test
_combined.hex
Parsing input hex files.
Merging files.
ERROR: The hex files cannot be merged since there are conflicts.`

Start of Primo hex (works):
`:020000021000EC
:10C000000000012095C40100D5C40100D7C401007F
:10C01000D9C40100DBC40100DDC401000000000040
:10C02000000000000000000000000000DFC401006C`

Start of Arduino Core hex (doesn't work):
`:100000000000012049050000710500007305000093
:10001000750500007705000079050000000000006C
:100020000000000000000000000000007B05000050
:100030007D050000000000007F0500008105000034
:100040008305000083050000C50A00008305000049
:100050008305000083050000830500008305000080`

When I look at the softdevice hex the conflict is obvious. How do I get Ardiuno Core hex in the right (higher value) sectors?

dlabun, 2017-05-16:
When compiling in the IDE do you have the soft device selected?

turinturam, 2017-05-16:
That was it. Oh man, its always the stupidest little things. Thanks for leading the way.

--
Curt White
Wearable Device Developer
Web Developer
Child Mind Institute
Office: 646.625.4268
Cell: 703.409.7471 (text first)

childmind.org<http://www.childmind.org/>  |  C<http://childmind.org/about-us/contact-us/>ontact<http://childmind.org/about-us/contact-us/>

Facebook<https://www.facebook.com/ChildMindInstitute>  |  Twitter<https://twitter.com/ChildMindDotOrg>

---

This email message, including any attachments, is for the sole use of the intended recipient(s) and may contain information that is proprietary, confidential, and exempt from disclosure under applicable law. Any unauthorized review, use, disclosure, or distribution is prohibited. If you have received this email in error please notify the sender by return email and delete the original message. Please note, the recipient should check this email and any attachments for the presence of viruses. The organization accepts no liability for any damage caused by any virus transmitted by this email.

dlabun, 2017-05-16:
Closing as resolved

closed 2017-05-16 by dlabun