sandeepmistry/arduino-nRF5 #164: ST Link v2 tries to write registers outside of Flash range (128kB nrf51822)

ptcoregon, 2017-05-24:
Hi there! Thanks for this great library. I'm having trouble flashing a program onto a generic nrf51822 (128kB) with an ST-Link v2. I can flash the softdevice with no problem, but the programs always fails verification. Here's the trace:

`Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0x4000
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.239604
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x000006d0 msp: 0x000007c0
** Programming Started **
auto erase enabled
Info : nRF51822-QFAB(build code: C0) 128kB Flash
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000001e msp: 0x000007c0
Warn : no flash bank found for address 20000
wrote 32768 bytes from file C:\Users\Peter\AppData\Local\Temp\arduino_build_115189/starter.ino.hex in 1.884715s (16.979 KiB/s)
** Programming Finished **
** Verify Started **
Error: timed out while waiting for target halted
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x81000003 pc: 0xfffffffe msp: 0x000007a0
Error: error executing cortex_m crc algorithm
Error: checksum mismatch - attempting binary compare
diff 0 address 0x00020020. Was 0x00 instead of 0x04
diff 1 address 0x00020021. Was 0x00 instead of 0x20
diff 2 address 0x00020023. Was 0x00 instead of 0x20
diff 3 address 0x00020024. Was 0x00 instead of 0x38
diff 4 address 0x00020025. Was 0x00 instead of 0xfd
diff 5 address 0x00020026. Was 0x00 instead of 0x01
diff 6 address 0x00020028. Was 0x00 instead of 0xdf
diff 7 address 0x00020029. Was 0x00 instead of 0xfb
diff 8 address 0x0002002a. Was 0x00 instead of 0x01
diff 9 address 0x0002002c. Was 0x00 instead of 0xd2
diff 10 address 0x0002002d. Was 0x00 instead of 0xfb
diff 11 address 0x0002002e. Was 0x00 instead of 0x01
diff 12 address 0x00020030. Was 0x00 instead of 0xd2
diff 13 address 0x00020031. Was 0x00 instead of 0xfb
diff 14 address 0x00020032. Was 0x00 instead of 0x01
diff 15 address 0x00020034. Was 0x00 instead of 0xd2
diff 16 address 0x00020035. Was 0x00 instead of 0xfb
diff 17 address 0x00020036. Was 0x00 instead of 0x01
diff 18 address 0x00020038. Was 0x00 instead of 0xd2
diff 19 address 0x00020039. Was 0x00 instead of 0xfb
diff 20 address 0x0002003a. Was 0x00 instead of 0x01
diff 21 address 0x0002003c. Was 0x00 instead of 0xd2
diff 22 address 0x0002003d. Was 0x00 instead of 0xfb
diff 23 address 0x0002003e. Was 0x00 instead of 0x01
diff 24 address 0x00020040. Was 0x00 instead of 0xd2
diff 25 address 0x00020041. Was 0x00 instead of 0xfb
diff 26 address 0x00020042. Was 0x00 instead of 0x01
diff 27 address 0x00020044. Was 0x00 instead of 0xd2
diff 28 address 0x00020045. Was 0x00 instead of 0xfb
diff 29 address 0x00020046. Was 0x00 instead of 0x01
diff 30 address 0x00020048. Was 0x00 instead of 0xd2
diff 31 address 0x00020049. Was 0x00 instead of 0xfb
diff 32 address 0x0002004a. Was 0x00 instead of 0x01
diff 33 address 0x0002004c. Was 0x00 instead of 0xd2
diff 34 address 0x0002004d. Was 0x00 instead of 0xfb
diff 35 address 0x0002004e. Was 0x00 instead of 0x01
diff 36 address 0x00020050. Was 0x00 instead of 0xff
diff 37 address 0x00020051. Was 0x00 instead of 0xff
diff 38 address 0x00020052. Was 0x00 instead of 0xff
diff 39 address 0x00020053. Was 0x00 instead of 0xff
diff 40 address 0x00020054. Was 0x00 instead of 0xff
diff 41 address 0x00020055. Was 0x00 instead of 0xff
diff 42 address 0x00020056. Was 0x00 instead of 0xff
diff 43 address 0x00020057. Was 0x00 instead of 0xff
diff 44 address 0x00020058. Was 0x00 instead of 0xff
diff 45 address 0x00020059. Was 0x00 instead of 0xff
diff 46 address 0x0002005a. Was 0x00 instead of 0xff
diff 47 address 0x0002005b. Was 0x00 instead of 0xff
diff 48 address 0x0002005c. Was 0x00 instead of 0xff
diff 49 address 0x0002005d. Was 0x00 instead of 0xff
diff 50 address 0x00020060. Was 0x00 instead of 0x39
diff 51 address 0x00020061. Was 0x00 instead of 0x81
diff 52 address 0x00020062. Was 0x00 instead of 0x01
diff 53 address 0x00020064. Was 0x00 instead of 0xe9
diff 54 address 0x00020065. Was 0x00 instead of 0x82
diff 55 address 0x00020066. Was 0x00 instead of 0x01
diff 56 address 0x00020068. Was 0x00 instead of 0x29
diff 57 address 0x00020069. Was 0x00 instead of 0x84
diff 58 address 0x0002006a. Was 0x00 instead of 0x01
diff 59 address 0x0002006c. Was 0x00 instead of 0xf9
diff 60 address 0x0002006d. Was 0x00 instead of 0xa7
diff 61 address 0x0002006e. Was 0x00 instead of 0x01
diff 62 address 0x00020070. Was 0x00 instead of 0x0d
diff 63 address 0x00020071. Was 0x00 instead of 0x81
diff 64 address 0x00020072. Was 0x00 instead of 0x01
No more differences found.
embedded:startup.tcl:454: Error: ** Verify Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 498
at file "embedded:startup.tcl", line 454
the selected serial port at file "embedded:startup.tcl", line 454
 does not exist or your board is not connected`

Since I'm using the 128kB flash verison of the nrf51822, it's trying to write outside the valid range. I can only select the 256kB version with your library, so is there something I need to do in order to make it compatible with my version?

Thanks for any help!

ptcoregon, 2017-05-25:
After a good sleep I figure out how to create a new board definition and get the program to verify after flash. However, the program still isn't functioning (even a simply blinky program).

I added this to the boards file:
Generic_nRF51822.menu.chip.xxxx=16 kB RAM, 128 kB flash (xxxx)
Generic_nRF51822.menu.chip.xxxx.build.chip=xxxx

And then created armgcc_s110_nrf51822_xxxx and changed the MEMORY to:
FLASH (rx) : ORIGIN = 0x0, LENGTH = 0x20000
RAM (rwx) :  ORIGIN = 0x20002000, LENGTH = 0x2000

I was able to change these values around and still get a successful flash, as long as the origin and length of flash did not exceed 0x20000 and the length was large enough for the hex file. For example, I also used FLASH (rx) : ORIGIN = 0x18000, LENGTH = 0x2000 with success as well as 
RAM (rwx) :  ORIGIN = 0x20000000, LENGTH = 0x4000

I selected my new board in the Arduino IDE, flashed the S110 Softdevice, and uploaded the program with the trace below:

Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0x4000
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.238020
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x000006d0 msp: 0x000007c0
** Programming Started **
auto erase enabled
Info : nRF51822-QFAB(build code: C0) 128kB Flash
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000001e msp: 0x000007c0
wrote 33792 bytes from file C:\Users\Peter\AppData\Local\Temp\arduino_build_975719/starter.ino.hex in 2.156900s (15.300 KiB/s)
** Programming Finished **
** Verify Started **
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x000007c0
verified 32888 bytes in 0.265206s (121.103 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked

But the program doesn't work correctly. Am I somehow having an issue with the SoftDevice after changing the flash around? Thanks for any suggestions!

ptcoregon, 2017-05-25:
It appears that my solution above _did_ flash the program correctly, but my LED was bad for blinky tests and my bluetooth test (starter.ino for BLE_peripheral) is hanging on begin(). I'll move my question over to that repo as this library appears to be working fine with my board definition fix.

closed 2017-05-25 by ptcoregon

qq597578205, 2017-09-11:
Is where created armgcc_s110_nrf51822_xxxx and changed the MEMORY?