sandeepmistry/arduino-nRF5 #165: Posibility to support smartwatches and blackmagicprobe

goran-mahovlic, 2017-05-26:
Is it ok with you if I push changes that add two smartwatches and Blackmagic probe as programmer?
I have tested it with your newest repo and initial tests are ok. Serial is working. SPI oled is working, and I can find I2C device on bus...

d00616, 2017-05-28:
@goran-mahovlic Hello Goran, Is there any place you have shared your findings?

goran-mahovlic, 2017-05-28:
We have gitter opened for some time now 
https://gitter.im/nRF51822-Arduino-Mbed-smart-watch/Lobby
some pics for ID107HR
https://www.facebook.com/mistery3/media_set?set=a.10154513788647158.1073741847.820257157&type=3
my blog post fot ID003 version
http://www.lemilica.com/smarty-the-watch-it-is-aliveeee/
 And for this version I have created new repo with libs for oled and for KX(still not working)
https://github.com/goran-mahovlic/nRF51_ID107_libs

I just used Rogers clone of this repo (he added watches and BMP).

dlabun, 2017-05-28:
Do you plan on selling the watches as a finished product or as a development kit?

goran-mahovlic, 2017-05-28:
I did not think of it that way that would take me lots of time, and  I have full time job this is only for hobby. 

I am interested of getting it running over DFU so everybody could just load code without opening and use it as they want.

And I am interesting in creating some opensource samples for that watch.
Few would be:
Midi instrument
Midi instrument for disabled people (we had one workshop for them,and original instruments are so expensive)
Epileptic seizure detection
Fall of an old person detection
Detecting to high hart rate.
...

And this is great development board for all of that except now it needs to be opened for loading samples :(

I will combine that watch with central nRF device - opensource mobile phone
http://www.lemilica.com/turning-cheap-gsm-development-board-into-opensource-ble-mobile-phone/
I will maybe if I got time make phone board ...

jacobrosenthal, 2017-05-28:
On Sun, May 28, 2017 at 5:41 AM, Goran Mahovlic <notifications@github.com>
wrote:

> I am interested of getting it running over DFU so everybody could just
> load code without opening and use it as they want.
>
> And I am interesting in creating some opensource samples for that watch.
> Few would be:
> Midi instrument
> Midi instrument for disabled people (we ha
>

Hah, Ive been reverse engineering id101
https://github.com/jacobrosenthal/arduino-nRF5/tree/id101

goran-mahovlic, 2017-05-28:
Join gitter chat, I will ask Roger to update his repo to newest so we can put ID101 there.
It already has ID003 and ID107HR 
https://github.com/rogerclarkmelbourne/arduino-nRF5
I can send you my my variant folder but you can try if it is working for you with this repo(there is possibility that ID101 has same pinout as ID107HR)
And with this repo you can try my libs for oled
And maybe find out why Wire lib is not working as it should...
In previews versions I had HRM sensor working to so when we get Wire to work we can focus on HRM...

curtpw, 2017-05-28:
I have the Si1143/Si1142 heart rate sensor working on the ID107 in mbed. Currently only using it for proximity though. It would be easy enough to take the raw photo diode values and plug chug them into a HR algo.

curtpw, 2017-05-28:
If anyone has the Si114X working on the ID107 Plus (nRF52832 http://www.idoosmart.com/ID107PlusHR/) please let me know. I'm working on it right now..... The ID107 Plus is a great little nRF52 dev board, I highly recommend it.

rogerclarkmelbourne, 2017-05-28:
@goran-mahovlic 
I'll try to update my repo later today

@curtpw 
I have a ID107HR  but not the ID107 Plus. What does the Plus have that the HR doesn't.

FYI.
Someone also told me about a H07 https://www.aliexpress.com/item/Bluetooth-Smart-Bracelet-Band-for-iPhone-Sync-Activity-Health-Tracker-Smartphone-Pedometer-Smart-Wirstband-Touch-Screen/32777549307.html

Which is supposed to have OTA capability, however I suspect they don't give out the key (but perhaps it is not secured OTA)

@dlabun 
The main problem with these devices (unless the ID107 Plus is any different), is that you need to dissasemble them and connect to a SWD programmer e.g. BMP or JLink or STLink etc, to reprogram them

It may be possible to upload via OTA to the H07 without disassembly, assuming its not secured, but it would still require users to transfer the binary to a mobile device to upload via Nordic's DFU toolbox etc.

As far as I can see the only way to make a this available to the general Arduino community would be to bundle the kit with a USB dongle in order to upload.
And so far I have not been able to get any normal USB BLE dongles e.g. the CSR ones, to connect to the nRF51 even on Windows 10 (and W7 does not have native support for BLE, and I'm not sure how well it works in W8 or W8.1)
The Nordic dongle could be made to do what's needed, but its very expensive, so would make the whole kit far to costly.

dlabun, 2017-05-28:
From the sound of it, including these watches wouldn't be a good fit with this repository since they're not intended for development usage.

rogerclarkmelbourne, 2017-05-29:
@goran-mahovlic 
BTW. I think @sandeepmistry  already said that he didn't want to add support for BMP

So, but I have this added to my fork already.

I just need to update my fork and the latest changes and re-integrate the smart watches into boards.txt

goran-mahovlic, 2017-06-05:
tnx, 
i will then probably try to make development board with same pinout as watch...

closed 2017-06-05 by goran-mahovlic

rogerclarkmelbourne, 2017-06-06:
@curtpw 

I'm thinking of buying a ID107 Plus

I'll send you an email to curt3n5@gmail.com

d00616, 2017-07-16:
@rogerclarkmelbourne You can manage boards outside of this repository. I think it simpler as managing a fork. At the moment I don't know how it works in detail, but you can find it in MySensors: https://github.com/mysensors/ArduinoBoards

rogerclarkmelbourne, 2017-07-17:
@d00616 

I tried adding the JSON url to my IDE prefs but nothing seemed to change in the Boards Manager

When you tried it, did some more options to install the MySensors boards appear ?

d00616, 2017-07-17:
@rogerclarkmelbourne Sorry, I can't help at the moment. I have seen this first time yesterday.

d00616, 2017-07-27:
@rogerclarkmelbourne If you follow the instructions of https://github.com/mysensors/ArduinoBoards you can install the My Sensors nRF5 boards. There are two boards allowing to define the pin layout in your sketch. If you want to define your own boards, you can start with https://github.com/mysensors/ArduinoHwNRF5 and look into the bpt files of the ArduinoBoards repository.

rogerclarkmelbourne, 2017-07-27:
Thanks

I've not had time to build a boards package yet, but I did look at the mysensors and I think I understand the concept and basics of how to do it.

micooke, 2017-08-21:
@dlabun -there are a few people working on the nrf5 based sport watches which are using this repo. Currently this requires @rogerclarkmelbourne to keep a separate outdated snapshot of this repo, just to support a couple extra boards (and BMP support).

As the nrf5 uses GPIO rather than dedicated pins, the pin mapping is crucial. So for this reason can i ask that these boards be supported? Im happy to action any issues that arise from the id107hr as i have one broken out as a dedicated development board.

sandeepmistry, 2017-09-05:
We might want to explore the smart watch core referencing this one instead of having a fork. This would apply if it only contains additional variants.

For example the Blend core references the official Arduino AVR core: https://github.com/RedBearLab/Blend/blob/master/Arduino/hardware/blend/avr/boards.txt

micooke, 2017-09-05:
Aah thanks for the reference!
Its probably a good idea to keep it separate as ive found  that the pinout is different for different versions of the same device.

micooke, 2017-09-06:
@rogerclarkmelbourne - FYI, this is pretty much a cut down version of your arduino-nRF5-customised repo.
@sandeepmistry - Well that was a pain!
https://micooke.github.io/package_nRF5_smartwatches_index.json

Which grabs an alpha release of [arduino-nRF5-smartwatches](https://github.com/micooke/arduino-nRF5-smartwatches)

The alpha doesn't define LED_BUILTIN and it replicates your platform.txt so doubles up on programmers. Ill tidy this up for the 0.1.0 release.

For reference, referencing a core of an installed package requires your board variants etc to be installed as a package also. 

Otherwise cloning your arduino-nRF5 and my Arduino-nRF5-smartwatches into your sketchbook hardware folder works.

sail2themoon, 2022-10-14:
Hi there,
I know this is an old thread, but I still decided to get a ID107 HR Plus, which is supported in the micooke-nRF5-boards v2.1. However, when I try to compile the example "U8g2_ScreenTest" or another example-sketch, I get: 
`/home/user/.arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file nrf52_{build.chip}.ld: File not found
collect2: error: ld returned 1 exit status
exit status 1
Error compiling Board ID107HR Plus.
`
But If I select as board the ID107 HR (not Plus), I'm able to compile the code. Does anyone know, what's going wrong here?

carlosperate, 2022-11-07:
Looks like the "ID107 HR Plus," board definition has not set the `build.chip` property:
https://github.com/micooke/arduino-nRF5-smartwatches/blob/defb6758ab3467e0512b48fc099fe35d757f9886/boards.txt#L135-L166

For example, the non-plus version has this in line 115:
```
ID107HR.build.chip=xxac
```
https://github.com/micooke/arduino-nRF5-smartwatches/blob/defb6758ab3467e0512b48fc099fe35d757f9886/boards.txt#L115

The equivalent is missing for the "Plus" version.

If it's the same chip as the other one (which I don't know, I'm not familiar with this watches at all) you could add this line manually in the "plus" section: `ID107HRPlus.build.chip=xxac`
But you'll have to figure out how to find the `boards.txt` file from the Arduino Core installation.