sandeepmistry/arduino-nRF5 #166: BLE Central examples for nRF51

netkruzer, 2017-06-10:
I see that nRF51 chips support Central and Peripheral modes in SoftDevice 130 v2.0.1 but I don't see any examples for a Central/Master BLE device that can connect to other nRF51 or nRF52 chips.

Does anyone have an example for BLE Central to scan and connect to other devices?

S3ler, 2017-06-13:
I am very interested in such an example, too.
Unfortunately I could not find any arduino-BLECentral library or similiar.
But: 
> All the relevant Nordic SoftDevice (S110, S130, S132) header files are included build path when a SoftDevice is selected

So it should be possible to use one of Nordic Semiconductor's central device examples as a starting point.

netkruzer, 2017-06-13:
Are you just able to run Nordic examples directly from the Arduino IDE without​ modification?

I need to give that a try.

S3ler, 2017-06-14:
Today I tried to run the Nordic example directly from the Arduino IDE and my result is:
Runs without modification? Clearly not! The examples use a lot of files from Nordic's SDK not included in the core/nRF5/SDK/components/softdevice/s130/headers folder. I even tried to build my own Arduino library including these files, but I could not make it work.

Now I am trying to build a Simple BLE Scanner example based on the [Heart Rate Monitor](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v13.0.0%2Fexamples.html) example from the [nRF5 SDK 12.3.0](https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF5-SDK)

I will update my comment when I have new results.

dlabun, 2017-06-15:
As far as I know Sandeep has not built a Central / Master version of his BLE library. This core is really only intended to get the basic features like SPI and I2C working, the Bluetooth master or ANT capabilities still require you to build your own library.

S3ler, 2017-06-15:
Ok, i tried to insert Nordic's SDK into the sandeepmistry/arduino-nRF5 repository (see my fork [S3ler/arduino-nRF5 ](https://github.com/S3ler/arduino-nRF5) ).
I documented my steps in [Steps.md](https://github.com/S3ler/arduino-nRF5/blob/master/Steps.md).

But now I am really stuck, I get the compilation error:
```
In file included from ~/.arduino15/packages/sandeepmistry/hardware/nRF5/0.3.0/cores/nRF5/SDK/components/libraries/atomic_fifo/app_atfifo.c:41:0:
~/.arduino15/packages/sandeepmistry/hardware/nRF5/0.3.0/cores/nRF5/SDK/components/libraries/atomic_fifo/app_atfifo_internal.h:61:9: note: #pragma message: value of __CORTEX_M: (0x00U)
 #pragma message "value of __CORTEX_M: " STR(__CORTEX_M)
         ^
~/.arduino15/packages/sandeepmistry/hardware/nRF5/0.3.0/cores/nRF5/SDK/components/libraries/atomic_fifo/app_atfifo_internal.h:62:2: error: #error Unsupported core version
 #error Unsupported core version
  ^
exit status 1
```
There are multiple checks like this:
```
#if ((__CORTEX_M >= 0x03U) || (__CORTEX_SC >= 300U)) == 0
#error Unsupported core version
#endif
```
But I do not have a clue how to improve the ```__CORTEX_M``` value. I am sure it is related with the ```gcc-arm-none-eabi toolchain``` an the ```-mcpu=cortex-m0``` compile flag.

Can someone help me?

dlabun, 2017-06-17:
This core is using an older version of the Nordic SDK, it's either 9 or 10... I am not 100% of the version but it's definitely not compatible with SDK v12.

sandeepmistry, 2017-06-18:
This should be possible with the older S130/S132 SoftDevice. There's some discussion on scanning here https://github.com/sandeepmistry/arduino-BLEPeripheral/pull/155 with @floe.

It would be nicer to see a `NordicBLE` library made that has the same API as the Arduino 101's CurieBLE: https://github.com/01org/corelibs-arduino101/tree/master/libraries/CurieBLE

S3ler, 2017-06-18:
I have now an overview over the Nordic SDK and used Eclipse CDT including a ST-Link V2 to program my Waveshare BLE400 board.
@dlabun Can we move to Nordic SDK version 12.3.0? This includes Softdevice S130 and is the last Nordic SDK version supporting nRF51822 (see [release notes](http://developer.nordicsemi.com/nRF5_SDK/nRF5_SDK_v13.x.x/doc/13.0.0/release_notes.txt)), so there should be further core migrations in the future?
@sandeepmistry Thank you for these links i will take a deeper look into them, especially the CureBLE API. Would you be ok with adding Nordic's SDK into the project (directly within git or via post install download script)? 
If yes: I can think of writing a wrapper around Nordic's SDK with the same API as the CurieBLE.

Give me your opinions on my idea.

ayavilevich, 2017-07-19:
Sandeep's library is very easy to work with and a nicely designed API, but as mentioned above it doesn't support Central roles or running Central at the same time as running Peripheral. I would like to see a library that is as easy to understand as Sandeep's but can support Central with many Peripherals and Central and Peripheral roles at the same time.

Naturally, it makes sense to get an initial version out quickly, even if it is only Peripheral. Thanks.

sandeepmistry, 2017-07-31:
@S3ler 

> Can we move to Nordic SDK version 12.3.0? This includes Softdevice S130 and is the last Nordic SDK version supporting nRF51822 (see release notes), so there should not be further core migrations in the future?

Please submit a pull request to do so.

> Would you be ok with adding Nordic's SDK into the project (directly within git or via post install download script)?

I don't think it is really needed, plus it has a non-open source license.

kriswiner, 2017-09-20:
Is someone planning to add the BLECentral component to Mistry's Arduino core so that we can use it just like we use BLEPeripheral? Not sure how to do this myself. Or maybe this is straightforward to use as is. I tried but had a lot of compile errors, etc. just from the fact I have no idea what I am doing.

Anyway, I would love to be able to use the central role examples in the Primo version of the nRF52 core with the core reposited here (which works for my custom nRF52 boards) if some knowledgeable person could explain how to do so. Thanks.

dmikhalsky, 2017-09-20:
I am totally with you on that. I tried many different approaches but nothing worked so it seems some cde alterations are needed.
As I can see. Primo's fork relies heavily on Nordic SDK

dlabun, 2017-09-20:
There was a pretty big discussion about this months ago (last year maybe?) and it was decided not to add BLE support directly to the core. Personally, I am not working on anything BLE related because I am working with the ANT related features.

dmikhalsky, 2017-09-20:
Will ANT implementation include Central role or whatever it's called in ANT specification?

dlabun, 2017-09-20:
Using ANT requires a license from Garmin so I won't be publishing my code related to ANT.

kriswiner, 2017-09-20:
I wasn't clear in my request. I agree we don't want BLE added to the nRF52
Arduino core.

What I am asking for is an API like BLEPeripheral that we could include in
core sketches but in this case it would be BLECentral and would allow
running the central examples in the Primo core.

I tried to do something like this but never got the examples to compile
because I was missing some Primo files or dependencies. This sounds
straightforward to fix for someone who knows the Mistry core and can more
properly formulate the BLECentral API from the Primo core so it can run
using the Mistry core.

Sadly, that person is not me.

tusharvb19, 2018-06-16:
Need BLE central and peripheral example to send and receive data.
Thanks in advance.

dlabun, 2018-09-03:
Closing due to lack of activity

closed 2018-09-03 by dlabun

TrailBee47, 2021-09-09:
last time I used nrf51 as central I used redbear labs library may be one can try that I will see if I could do it. with Sandeep's core. it has same nrf51 support and ble examples for central. https://github.com/RedBearLab/nRF51822-Arduino/tree/S130/arduino-1.6.x/hardware/RBL/RBL_nRF51822/libraries/BLE_Examples/examples/BLE_CentralTest