sandeepmistry/arduino-nRF5 #167: nRF5FlashSoftDevice support adafruit bluefruit nrf51822

claudioarduino, 2017-06-14:
@sandeepmistry  Hi sandeep, I'm working on adafruit bluefruit nrf51822 with arduino mega2560. I'm trying to flash the S130 soft device, because it supports both central and peripheral role, but it doesn't work. Is it possible to do? Does nRF5FlashSoftDevice support adafruit nrf51822?

Thanks for your help.
Claudio

S3ler, 2017-06-14:
Which board do use exactly? Most Adafruit Boards do not work, see Readme.md:
> WARNING: Adafruit nRF51 boards are NOT compatible with this library, this includes the Bluefruit LE UART Friend, Bluefruit LE SPI Friend, Bluefruit LE Shield, Flora Wearable Bluefruit LE Module, Feather 32u4 Bluefruit LE, and Feather M0 Bluefruit LE.

dlabun, 2017-06-15:
Sorry, it's not compatible with Adafruit products. I grabbed the info below off the Adafruit website at:
https://learn.adafruit.com/adafruit-bluefruit-le-shield/faq

-Can my Bluefruit LE module connect to other Bluefruit LE peripherals?
No, the Bluefruit LE firmware from Adafruit is currently peripheral only, and doesn't run in Central mode, which would cause the module to behave similar to your mobile phone or BLE enabled laptop.
If you required Central support, you should look at the newer nRF52832 based products like the Adafruit Feather nRF52 Bluefruit LE, which contains a SoftDevice which is capable of running in either Central or Peripheral mode. The nRF518322 based products (such as the one used in this learning guide) are not capable of running in Central mode because it isn't supported by the SoftDevice they use, and it isn't possible to update the SoftDevice safely without special hardware.

sandeepmistry, 2017-06-18:
@claudioarduino you'll have to use an external ST-LINK, JLINK, or CMSIS-DAP programmer to flash those boards. Then the Bluefruit board will run it's own Arduino core and you'll need some custom protocol between the Mega and board.

closed 2017-06-18 by sandeepmistry