sandeepmistry/arduino-nRF5 #168: Porting low power mode from Primo

curtpw, 2017-06-18:
Apparently there is now a low power library for the Primo (nrf52) from the folks at arduino.org. Not sure how difficult it would be to get that working with this repo: https://github.com/arduino-libraries/ArduinoLowPower/blob/master/examples/PrimoDeepSleep/PrimoDeepSleep.ino

If anyone is interested in porting or could point me in the right direction in terms of doing it myself it would be much appreciated.

dlabun, 2017-06-23:
It doesn't look like the library needs to be ported, it seems to support any generic nRF52 chip OR the Primo specifically since it uses a companion chip. The only thing the library doesn't have is an example for non-Primo nRF52 boards.

sandeepmistry, 2017-07-30:
@curtpw any updates on your efforts for this?

curtpw, 2017-08-02:
No, it needs a variety of dependencies. nrf_asset and a couple others. I guess the Primo is drawing on the/a NRF5 SDK in a fashion that distinguishes for your project. It would be nice if I could get it working. I might take another stab.

sandeepmistry, 2017-08-11:
@curtpw ok, older versions of the Nordic SDK had a weird license - I haven't checked newer versions. It would be awesome to get something done that just used the registers directly.

kriswiner, 2017-08-22:
Ran with the ArduinoLowPower library this sketch:

`/*
  TimedWakeup
  This sketch demonstrates the usage of Internal Interrupts to wakeup a chip in sleep mode.
  Sleep modes allow a significant drop in the power usage of a board while it does nothing waiting for an event to happen. Battery powered application can take advantage of these modes to enhance battery life significantly.
  In this sketch, the internal RTC will wake up the processor every 2 seconds.
  Please note that, if the processor is sleeping, a new sketch can't be uploaded. To overcome this, manually reset the board (usually with a single or double tap to the RESET button)
  This example code is in the public domain.
*/

#include "ArduinoLowPower.h"

ArduinoLowPowerClass LowPower;

void setup() {
  pinMode(7, OUTPUT);
  // Uncomment this function if you wish to attach function dummy when RTC wakes up the chip
  // LowPower.attachInterruptWakeup(RTC_ALARM_WAKEUP, dummy, CHANGE);
}

void loop() {
  digitalWrite(7, HIGH);
  delay(500);
  digitalWrite(7, LOW);
  delay(500);
  // Triggers a 2000 ms sleep (the device will be woken up only by the registered wakeup sources and by internal RTC)
  // The power consumption of the chip will drop consistently
  LowPower.sleep(10);
}

void dummy() {
  // This function will be called once on device wakeup
  // You can do some little operations here (like changing variables which will be used in the loop)
  // Remember to avoid calling delay() and long running functions since this functions executes in interrupt context
}`

I had to comment out a line in the .h file to get it to compile (using the nRF52 DK variant). 

      void sleep(void);
      //    void sleep(uint32_t millis);
      void sleep(int millis) {
      sleep((uint32_t)millis);
       }I

Good news is I did measure a 5 mA drop in power consumption, so this works, the bad news is the nRF52 never woke from sleep and the led remained on ;>

There is something wrong with the wakeup from sleep (at least with the nRF52 Arduino core). I couldn't get the wake from GPIO to even compile.

I hope someone who is a better C++ programmer than I am will take a look at making this library work with the nRF52 Arduino core. It would be very useful to cut power usage from 6 mA to ~1 mA!

curtpw, 2017-08-29:
I've been looking at the way sleep/low power modes are handled on mbed. I think this may provide some insight into how the Primo ArduinoLowPower library can be made to work. Specifically disabling RTOS on mbed in the context of sd_app_evt_wait()/NRF_POWER->SYSTEMOFF , sd_power_mode_set(NRF_POWER_MODE_LOWPWR) and enabling the DC to DC regulator ( sd_power_dcdc_mode_set(NRF_POWER_DCDC_ENABLE) ). I've been looking through code here:

https://developer.mbed.org/questions/77180/using-power-management-API-such-as-sd_ap/
https://devzone.nordicsemi.com/question/113487/nrf52-sleep-mode-and-ble-advertising-mbed/
https://vilimpoc.org/blog/2017/04/24/power-profiling-on-mbed-nordic-nrf5-series/

gtalusan, 2017-09-17:
I have a Redbear BLE Nano V2 and it consumes about 9.5-11mA.  I was poking around how to get it to consume less and I came across this issue.

I copied over the missing mbed SDK dependencies into the core and massaged it enough such that ArduinoLowPower compiled, but my board would never come out of sleep with either GPIO or timed wakeups.

kriswiner, 2017-09-17:
Forget the Low Power API, it doesn't work for non-Prino boards.

Use sd calls instead, to enable low power mode and use __WFE at the end of
your loop to allow sleep until an interrupt.

d00616, 2017-10-06:
@sandeepmistry Nordic has changed the licensing starting with SDK 14 the old license is replaced for major parts. Maybe chapter 4 is a little bit problematic.

> Copyright (c) 2010 - 2017, Nordic Semiconductor ASA
> 
> All rights reserved.
> 
> Redistribution and use in source and binary forms, with or without modification,
> are permitted provided that the following conditions are met:
> 
> 1. Redistributions of source code must retain the above copyright notice, this
>    list of conditions and the following disclaimer.
> 
> 2. Redistributions in binary form, except as embedded into a Nordic
>    Semiconductor ASA integrated circuit in a product or a software update for
>    such product, must reproduce the above copyright notice, this list of
>    conditions and the following disclaimer in the documentation and/or other
>    materials provided with the distribution.
> 
> 3. Neither the name of Nordic Semiconductor ASA nor the names of its
>    contributors may be used to endorse or promote products derived from this
>    software without specific prior written permission.
> 
> 4. This software, with or without modification, must only be used with a
>    Nordic Semiconductor ASA integrated circuit.
> 
> 5. Any software provided in binary form under this license must not be reverse
>    engineered, decompiled, modified and/or disassembled.
> 
> THIS SOFTWARE IS PROVIDED BY NORDIC SEMICONDUCTOR ASA "AS IS" AND ANY EXPRESS
> OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES
> OF MERCHANTABILITY, NONINFRINGEMENT, AND FITNESS FOR A PARTICULAR PURPOSE ARE
> DISCLAIMED. IN NO EVENT SHALL NORDIC SEMICONDUCTOR ASA OR CONTRIBUTORS BE
> LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
> CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE
> GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION)
> HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
> LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT
> OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
>

dlabun, 2017-10-06:
Chapter 4 shouldn't be an issue, we're only targeting nRF51 and nRF52 chips.

mristau, 2018-02-12:
Is there anybody who could tell me how to send the nRF52 to deep sleep/low power mode
how would i use the sd calls, are they available in this library?

dlabun, 2018-02-12:
Please read this issue, all of your questions are answered above.

probonopd, 2018-02-17:
> Please read this issue, all of your questions are answered above.

Maybe the information is hidden in this thread, but I would also appreciate an example sketch that shows how to send the nRF5 to deep sleep/low power mode in a way so that it wakes up again, e.g., after some time has passed and/or an input has triggered.

mristau, 2018-02-18:
I use this code to send it to power off

`void power_off() {
  digitalWrite(LED_BUILTIN, LOW);
  delay(200);
  
  uint8_t check;
  sd_softdevice_is_enabled(&check);

  if (check == 1) {
    Serial.println("SD Power OFF");
    sd_power_system_off();
  } else {
    Serial.println("NRF Power OFF");
    NRF_POWER->SYSTEMOFF = 1;
  }
}`

LED part is not important, but we have it blinking and it should be out, prints only for debug

`#include <nrf_sdm.h>
#include <nrf_soc.h>
#include <WInterrupts.h>`

using these includes, i don't remember exactly which were needed for the power off.

To get it to wake up again, i first attach an interrupt and then change the Sense_Config

`  attachInterrupt(digitalPinToInterrupt(LIS2DE12_INT1_PIN), int1Handler, RISING);
  NRF_GPIO->PIN_CNF[LIS2DE12_INT1_PIN] |= ((uint32_t)GPIO_PIN_CNF_SENSE_High << GPIO_PIN_CNF_SENSE_Pos);`

The code is from nrf5x_bitfields.h and wiring_digital.c

probonopd, 2018-02-18:
Thanks @mristau, very useful. Do you think you could make a small example sketch from it? I think this is to interest to many, if not all, users. BLE is about "low power", after all. (I am lacking the proper current measurement circuitry at the moment so I'd have to work "blindly".)

mristau, 2018-02-22:
i threw my code for the nRF lowPower in a small library
https://github.com/mristau/Arduino_nRF5x_lowPower

I tested most of it, but the DCDC Stuff, as it isn't in use on my board

kriswiner, 2018-02-22:
Excellent, thank you for posting. I'll try and report findings.

curtpw, 2018-02-22:
This great, thanks for working on this

sandeepmistry, 2018-02-26:
@mristau cool! If feedback on the library is positive, happy to merge a PR to add a link to the read me for a new "Recommended Libraries" section.

knopserl, 2018-12-09:
@mristau Thnaks for the lib but its a.) not compatible with the LowPower lib (which is also used for many otehr projects, so the API's shall be comparable) and b.) the most important part is the sleep(int ms). A sleep mathod. Thus I can't really use it becaus it seem to be only usable for external HW interrupts.

AliMalik5179, 2020-11-26:
Have you found any solution? I have tried @mristau Library but not able to awake Nrf52 Dk from deep sleep mode.

Mboudali, 2024-06-14:
up.