sandeepmistry/arduino-nRF5 #170: Using the bbc micro:bit display

lorenzoromagnoli, 2017-06-23:
It took me a while to understand how to use the display of the Arduino micro:bit with this core. 
It is not clear how the row/column pins of the Charlieplexed matrix are mapped. 

After a bit of trial and error, I could spot them all. 

```arduino
#define ROW_1  26
#define ROW_2  27
#define ROW_3  28

#define COL_1  3
#define COL_2 4
#define COL_3  10
#define COL_4  23
#define COL_5  24
#define COL_6  25
#define COL_7  9
#define COL_8  7
#define COL_9  6
```
I'm not sure if this is an expected result or not. 
By looking at the [micro:bit schematics](https://github.com/bbcmicrobit/hardware/blob/master/SCH_BBC-Microbit_V1.3B.pdf) infact, I would have expected to use different pins to switch the matrix:

```
row_pins ={ 19, 20, 21}
col_pins ={8,9,10,11,12,13,14,15,16,17,18}
```
[image]

I believe there is something wrong in the [variant.cpp](https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/BBCmicrobit/variant.cpp) file, but I'm not really sure how this is used. 

Happy to contribute with a PR after a bit of clarification :)

Meanwhile, I'm working on a little library to control the[ micro:bit display](https://github.com/lorenzoromagnoli/Microbit_display);  before tagging the first release I would like to make sure that the pins I'm using are the correct one.

❤️ Thanks for this Core ❤️

dlabun, 2017-06-23:
It looks like everything is behaving as expected. The variant.cpp file is basically a translation from the hardware port numbers to an Arduino user friendly pin number. What might be confusing is that we're going by the hardware port numbers, which all start with "P0." and we don't use the full port in number in the file.

Here's the first 5 pins listed in variant.cpp written slightly differently so its easier to see what's happening

P0.03 = Arduino Pin 0  <-- PAD1
P0.02 = Arduino  Pin 1 <-- PAD2
P0.01 = Arduino Pin 2  <-- PAD3
P0.04 = Arudino Pin 3  <-- COL1
P0.05 = Arduino Pin 4  <--COL2

lorenzoromagnoli, 2017-06-23:
Ok, got it! 
for clearance, it might be useful explain this in a README.

sandeepmistry, 2017-07-30:
@dlabun thanks for helping out!

@lorenzoromagnoli maybe you can start a Wiki page for the Micro:bit?

https://github.com/sandeepmistry/arduino-nRF5/wiki

I think the read me is not a good place for micro:bit specific info. Thoughts?

(closing for now, but let's continue the discussion here.

closed 2017-07-30 by sandeepmistry