sandeepmistry/arduino-nRF5 #171: I2S

kriswiner, 2017-06-27:
I'd like to use the Arduino I2S library (ArduinoSound) but I am not sure how to specify the I2S pins using the generic nRF52 board variant, nor which pins are "standard" for I2S in the Arduino Sound library. Is there an example of how to use an I2S microphone with the nRF52 Arduino core?

sandeepmistry, 2017-07-31:
@kriswiner the I2S library has now yet been ported over. Pull requests welcome ...

Protonerd, 2018-10-23:
Hi Sandeep, I'm also planning to use the nRF52 (more specifically the MDBT42Q module) on a custom project and I need to drive an I2S audio amp with it. Most ARM based cores have a fixed pinout, assigning functionalities to pins in a hard way (like SAMD21 etc). But it seems to be different for the nRF52. So what would be the right way to connect an I2S module, like the MAX98357A ?