sandeepmistry/arduino-nRF5 #172: J Link is not working

fabiencomte, 2017-07-02:
Hello,
I tried a basic exemple with an nRF52832 and a Segger Jlink probe and it not works. This is the output is says that JTAG is not configured but my probe is well installed (i use it with others targets). Do you have an idea please ?

...
Sketch uses 2524 bytes (0%) of program storage space. Maximum is 409600 bytes.
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
0
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
jaylink: Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Error: No J-Link device found.
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.

dlabun, 2017-07-02:
Did you follow the setup directions for the J-Link in the read me file?

fabiencomte, 2017-07-02:
Hello,
I re did the procedure and output changed for this one. Thank you for your help.

Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
0
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
in procedure 'program' 
in procedure 'init' called at file "embedded:startup.tcl", line 473
in procedure 'ocd_bouncer'
** OpenOCD init failed **
shutdown command invoked

An error occurred while uploading the sketch

fabiencomte, 2017-07-02:
If i erase all the cpu, it looks like if it was fixed but my program do Nothing (a simple blink a LED program)

fabiencomte, 2017-07-02:
I found, softdevice hex was missing and i burned bootloader in arduino.
Thank you for your help.

closed 2017-07-02 by fabiencomte