sandeepmistry/arduino-nRF5 #173: changing uart pins

oukradation, 2017-07-02:
Hi,

First of all thanks a lot for putting this up. It made it a lot easier to start with nrf52 chips for a beginner like me coming directly from Arduino. 

In my project, I have been trying to set up multiple Uart connection by changing pins in right time. So far I was able to instantiate 2 different Uart pins setups and able to print on both but not able to read.

In my code it goes something like:

#define RX_BUFFER_SIZE 256

Uart device1( NRF_UART0, UARTE0_UART0_IRQn, 24, 25 );
Uart usb ( NRF_UART0, UARTE0_UART0_IRQn, 22, 23 );

char rx_buffer[RX_BUFFER_SIZE];

void uart_transmit(Uart *uart, char *buff){
  uart->begin(9600);
  uart->println(buff);
  uart->end();
}

void uart_transmit_receive(Uart *uart, char *buff){
    uart->begin(9600);
    uart->print(buff);
    memset(buff, ' ', RX_BUFFER_SIZE);
    buff[0]='\0';
    delay(30);
    int i = 0;
    while(1){
      if(uart->available()>0){
        buff[i++] = (char)uart->read();
      }
      if(i>RX_BUFFER_SIZE || (buff[i-2] == 'O' && buff[i-1] == 'K')
          || (buff[i-2] == 'O' && buff[i-1] == 'R')) {
            buff[i] = '\0';
            break;
          }
    }
    
    delay(30);
    uart->end();
}

I see that each uart connections are printing out but uart->available() seems to return 0 all the time. Which probably indicate that interrupt is not working correctly, but I have no idea how I should approach this. If I change "uart->" to default "Serial." it works fine on default uart pins. I am working on Sparkfun nRF52832 breakout board on this project. 

In SDK provided by Nordic semiconductors I was able to achieve something similar with app_uart_init() and app_uart_uninit() on nRF52 DK. 

Any advice would be very much appreciated. Thank you.

dlabun, 2017-07-03:
The nRF chips only have a single UART so it sounds like you're trying to implement a software UART... I would recommend you ask over on the Nordic Semiconductor for input as you're probably not going to get a fast answer here.

oukradation, 2017-07-03:
Hi, 

yes, I understand that nRF52832 features only one uart, but since any GPIO can be used for uart, I was planning to channel between different devices instead of having external multiplexer.

dlabun, 2017-07-03:
That's something to ask Nordic about, their engineers would be able to tell you if the chip is capable of doing what you're asking for.

mlu, 2017-07-08:
There is only one physical UART so all three Serial, device1 and usb will work with the same hardware registers but using different pin configurations. The IRQ is working only with the standard Serial device datastructure. What might work is to configure pins and baudrates with:

`uart->begin(9600);`
Then use Serial to do the actual read and write without any begin(), and end that 'session' with 
`uart->end();` and perhaps also a `Serial->end();` to clear the Serial rxBuffer.

This is a bit messy, a better way would be to add a SetRxTxPins() method to the UART class.

sandeepmistry, 2017-07-31:
@oukradation there are no plans to submit changing the UART pins at runtime. Mainly because the official Arduino API's do not support this. Feel free to hack the core to gain this functionality though :)

closed 2017-07-31 by sandeepmistry