sandeepmistry/arduino-nRF5 #174: Pin interrupts an timers

fabiencomte, 2017-07-03:
Hello,

Is library support interrupts on inputs and timers ? If yes, where can i find an exemple please ?

Thank you

kriswiner, 2017-07-04:
I am using interrupts in this sketch:

https://github.com/kriswiner/nRF52832DevBoard

Is that your question?

fabiencomte, 2017-07-07:
Hello,
Thank you for your response. I was more thinking about GPIO input and timer tick every 50 ms. Do you have any tips ?

kriswiner, 2017-07-07:
What is the difference between GPIO input and an interrupt? I don't
understand what you want to do...

As far as getting a timer tick every 50 ms, or better yet, getting an
interrupt at a timed interval, I don't know how to do this with the nRF52.
It has no RTC or RTC alarm like an STM32L4. I don't think there is a timer
class available in the Arduino IDE. Maybe you could port one from the
Nordic SDK?

fabiencomte, 2017-07-07:
I found handlers about BLE in your link, maybe i missed something ?

For timer, there are some inside. I know how to use them in SDK way (I use them in other projects) but not "Arduino" style.

kriswiner, 2017-07-07:
"I found handlers about BLE in your link, maybe i missed something ?"

Can you say more clearly what you are looking for? I still do not
understand what the problem is.

fabiencomte, 2017-07-07:
I wish to use interrupts for GPIO inputs and timers and I am searching a document or an exemple how to do that.

fabiencomte, 2017-07-07:
For now I always used Nordic SDK with nRF52 (see https://tmblr.co/ZkUPrk238-oaF) but it's not very user friendly and I want to do a try with Arduino for a fast small proof of concept.

kriswiner, 2017-07-07:
OK, I shared an example of using an interrupt from a sensor. Is this not
what you mean by "interrupts for GPIO inputs"?

You will find very little documentation about the nRF52 Arduino core. I
have some examples that you might be able to sue.

sandeepmistry, 2017-07-31:
@fabiencomte timers are currently not supported.

closed 2017-07-31 by sandeepmistry