sandeepmistry/arduino-nRF5 #178: support for the nRF52 Blend v2

gesaleh, 2017-07-22:
Hi
does this library support the new device from RedBear ? blend 2 nano2 ?
I find your work very important and make it easy to have peripheral devices.
Thanks

gesaleh, 2017-07-22:
ok it wasn't that the issue it is the TimerOne lib issue
I can't compile it I'm getting an error al the time
`temp_sensor:140: error: 'class TimerOne' has no member named 'initialize'

   Timer1.initialize(2 * 1000000); // in milliseconds

          ^

temp_sensor:141: error: 'class TimerOne' has no member named 'attachInterrupt'

   Timer1.attachInterrupt(timerHandler);

          ^

exit status 1
'class TimerOne' has no member named 'initialize'`

dlabun, 2017-07-23:
Yes, both the Redbear Blend 2 and Nano 2 are supported.

sandeepmistry, 2017-07-31:
Please see answer here: https://github.com/sandeepmistry/arduino-BLEPeripheral/issues/169#issuecomment-317247667

closed 2017-07-31 by sandeepmistry