sandeepmistry/arduino-nRF5 #180: nfr51822 i2c incorrect readings help request

OracFire, 2017-08-01:
I need help with incorrect readings of I2C devices using generic nfr51822 boards.
Using standard ATmega328P Arduino I can read a DS3231 real time clock.

2017/7/31 (Monday) 16:48:5 since midnight 1/1/1970 = 1501519685s = 17378d
2017/7/31 (Monday) 16:48:6 since midnight 1/1/1970 = 1501519686s = 17378d

When using the same example on various nrf51822 generic boards running sandeepmistry Arduino Core I get incorrect values and the seconds end in a '7' and the minutes tick by the second.

2007/31/1 (Friday) 38:22:17 since midnight 1/1/1970 = 1200147737s = 13890d
2007/31/1 (Friday) 38:23:17 since midnight 1/1/1970 = 1200147797s = 13890d
2007/31/1 (Friday) 38:24:17 since midnight 1/1/1970 = 1200147857s = 13890d

I also get wrong data from an MLX90614 temperature sensor, and tried various nrf51822 boards, and with different DS3231 and MLX90614 Arduino libraries. I can drive an SSD1306 oled from the nrf51822 over i2c without problems. I have lots of Arduino experience, also C++ and some i2c DSO analysis.

Currently the common thread with these issues appears to be your excellent nrf51822 Arduino Core.
Does you know what could be wrong or can advise on what direction to learn about so I can fix this?

rogerclarkmelbourne, 2017-08-01:
Have you attached a logic analyser and investigated the differences in data between the AVR and nRF MCU's data?

I have raised an issue where the I2C seems to generate extra clocks e.g. if I request 2 bytes, I get 3 bytes worth of clocks (on the nRF52). 
I doubt you have the same bug, but its worth investigating with a logic analyser

PS. You can make a 16 channel logic analyser for less than $10 and use the free sigrok pulseview software
see https://www.youtube.com/watch?v=dobU-b0_L1I

OracFire, 2017-08-02:
I now have a logic analyser on order with results to follow. thanks

rmtucker, 2017-08-08:
Have you made sure that you have installed the development version from git as the instructions.
I had this problem until i updated.

sandeepmistry, 2017-08-11:
@OracFire this might be fixed in the master version as discussed in https://github.com/sandeepmistry/arduino-nRF5/issues/182#issue-248684630 and by @rmtucker above. Would you be able to try it out, instructions can be found here: https://github.com/sandeepmistry/arduino-nRF5#from-git-for-core-development.

RickyGer, 2017-08-15:
I have the same problem as OracFire. When measuring Arduino_Due + I2C + SHT30 everything works perfectly, but the sensor values from combination BLE400_Eval_Kit(NRF51822) + I2C + SHT30 are wrong.
The measurement with Sensor BME280 has the same problem.

Here are two oscillography:
1. Measurement from Arduino Due
2. Measure from the BLE400_Eval_Kit

[image]
[image]

The I2C signal is distorted by NRF51822, so the errors are generated.
Is there a solution to this?

Hardware:
BLE400 NRF51822 Eval Kit
Temperature Humidity Sensor Sensirion SHT30 (10K Pullups) or 
Temperature Humidity Sensor Bosch BME280 (4.3K Pullups) 

Software:
Arduino 1.8.3
arduino-nRF5-master (V 0.3.0) 
arduino-sht-master (for SHT30)
SparkFun_BME280_Arduino_Library-master (for BME280)

rogerclarkmelbourne, 2017-08-15:
This has nothing to do with software.

Its all a function of the hardware.

You should post your question to the Nordic Devzone forum, as its Nordic make the hardware

RickyGer, 2017-08-16:
The problem is solved!
Thanks to Roger for the quick response.
It was nevertheless a software problem, I have updated to the current version of the Sandeep Mistry "arduino-nRF5-master" and the problem of the wrong readings from the I2C interface is thus solved.
Greetings from Germany.

rogerclarkmelbourne, 2017-08-17:
You mean the software was causing the noisy signals ?

RickyGer, 2017-08-17:
Yes, absolutely this was a Software Bug. At the beginning I thought it would be a coupling or termination problem but then, I have compared the files from my local "arduino-nRF5-master" library (downloaded at the. 29 Jun 2017) to the actual one from the GitHub and noticed that there are a lot of changes in the TWI driver, so I updated that, and that solved the problem.

sandeepmistry, 2017-08-20:
Closing this, it should be resolved by the 0.4.0 release.

closed 2017-08-20 by sandeepmistry