sandeepmistry/arduino-nRF5 #182: Please crate a new arduino-nRF5 release

d00616, 2017-08-08:
@sandeepmistry @dlabun, Please publish a new release based on the actual development snapshot. There are open issues like https://github.com/sandeepmistry/arduino-nRF5/issues/180 which are already fixed.

This great Arduino port is the base for MySensors with nRF5 MCU's. Working I2C is mandatory.

Thanks.

dlabun, 2017-08-08:
Let's see what @sandeepmistry thinks about pushing a new release. If we don't hear from him in the next few day's I'll go ahead and create a new release.

sandeepmistry, 2017-08-11:
@dlabun it would be nice to get #175 in before doing a new release.

However, if we don't hear any updates in the next day or two, it would be awesome if you make a new `v0.4.0` release. The process will be similar to https://github.com/sandeepmistry/arduino-nRF5/commit/2913b153cd8fc5383f3d9517b0577bc6cd6ffb04 and https://github.com/sandeepmistry/arduino-nRF5/commit/fd5b88e55c0002c503db02a9fa0440e37d764684 (two commits required, first one to determine the checksum, size etc after the tag is made). Thanks!

dlabun, 2017-08-11:
@sandeepmistry Okay, I'll plan on Sunday morning reviewing everything to see if there's been any updates, otherwise I'll make a new release

sandeepmistry, 2017-08-12:
@dlabun sounds good, as discussed in #184, maybe we should hold off up I2C stop bit = false issues on the nRF51 are sorted out first? I'll be on the road for most of this weekend, but can set some time on the weekdays to take a closer look if needed.

dlabun, 2017-08-12:
@sandeepmistry Okay, I will hold off until next weekend pending resolution of the stop bit issue. If it's resolved sooner I will create the new release sooner.

sandeepmistry, 2017-08-18:
@dlabun I've tested and merged in #184.

dlabun, 2017-08-18:
@sandeepmistry I will publish the new release on Sunday assuming no critical issues are opened before then.

sandeepmistry, 2017-08-19:
@dlabun perfect!

dlabun, 2017-08-20:
v0.4.0 is now live, only took 3 commits.... I didn't realize Mac OS partials unzips tar.gz files so the MD5 checksum was wrong at first

sandeepmistry, 2017-08-20:
@dlabun awesome! Congrats on being the 1st person besides me to make a release!

Nice to see the Travis CI caught the checksum issue. Do you have any suggestions on how we could improve the process?

@d00616 I'm closing this as a new release is now available.

closed 2017-08-20 by sandeepmistry

d00616, 2017-08-20:
Thanks a lot.

dlabun, 2017-08-20:
@sandeepmistry The whole time I was doing the release I couldn't stop thinking why doesn't Github have a prepare an Arduino release feature? Every library and core uses the same JSON specs... Might be something to suggest to Github and the Arduino folks.

sandeepmistry, 2017-09-05:
@dlabun we could set something up using Travis CI or equivalent to do it when a new tag is made.

Most other cores don't host everything on Github, so not sure it would be useful for everyone. I know the official Arduino.cc JSON is hand edited though.